verilator/verilator #2432: question mark operator with a bit constant as condition gives potential parse issues

MaxStekelenburg, 2020-06-18:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
1'b1?4'hF:4'h1 parsed incorrectly (it takes 1'b1?4 together and then indicates the 4 is illegal)

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 4.034 2020-05-03 rev v4.032-127-g9dc65df98

Would you be willing to try to fix Verilator yourself with assistance?
Maybe but not sure if fixing this will not require to know the state of the parser as in some cases a '?' is a legal wildcard.

wallento, 2020-06-20:
Thanks for the report. Confirmed, I have pushed https://github.com/verilator/verilator/tree/issue-2432 to be fixed :)

wsnyder, 2020-06-28:
Verilator parses numbers in the lex stage as this code predates the IEEE standard. The lexer should instead only look for normal numbers and the grammar should then build the numbers appropriately as the IEEE grammar suggests.

The workaround is to add some appropriate spaces.

wsnyder, 2020-06-29:
Looking at this further I do not believe this code is guaranteed as legal according to IEEE.  Basically it depends on what the precedence is of ? versus numbers. My reading of the standard is that the ? is much lower than the number, but is potentially  ambiguous.

One major tool takes this.  Two other major tools give syntax errors, as does Verilator, so I'm sticking with that interpretation.

I've added this test case from the branch into t_number_bad.v so we have it for the future.

closed 2020-06-29 by wsnyder

wallento, 2020-06-29:
Reminds me of not only testing with this one that takes it :)

wsnyder, 2020-06-29:
@wallento, no worries, I thought it was legit also until looked at fixing it.