verilator/verilator #2433: Verilator warnings

agrobman, 2020-06-18:
Hi,

as response to 
https://github.com/chipsalliance/Cores-SweRV-EL2/pull/

we had to block WIDTH and UNOPTFLAT warnings as we used verilator 4.020 for cores release.
Seems majority of warnings are not helpful or incorrect. 

What is the current verilator "production" version?

Was these warnings detection improved?

Here are examples for EL2

```
%Warning-WIDTH: ... design/elx2_pic_ctrl.sv:412: Operator LT expects 32 bits on the RHS, but RHS's SEL generates 9 bits.
                                                                                                          : ... In instance tb_top.rvtop.swerv.pic_ctrl_inst
         for (int i=0; i<pt.PIC_TOTAL_INT_PLUS1; i++) begin

```

here we use sized parameters - the tool complains about size difference between parameter constant and integer . I would expect warnings/errors only in case where the constant does not fit a variable or   unsigned variable  gets/compared with negative constant.

```
Warning-WIDTH: ... estbench/tb_top.sv:348: Operator ASSIGNDLY expects 1 bits on the Assign RHS, but Assign RHS's REPLICATE generates 2 bits.
                                                                                                      : ... In instance tb_top
        wb_valid  <= {wb_valid[1],rvtop.swerv.dec.dec_i0_wen_r};

```

wb_valid is array of two bits : logic wb_valid[2]

wallento, 2020-06-20:
This is a good argument to use waivers, as it gives you the ability to filter those warnings that are either too pedantic or bogus, which is by the way the content of the PR you are "replying" to.

@wsnyder I also had a look and EL2 is a codebase that has many of those WIDTH rule warnings. I have previously worked on #2070 and found that there are probably quite a couple of strange cases hidden in how `V3Width` currently works. What are your experiences with it? To me it seems we should have a closer look and collect strange and seemingly overly pedantic warnings either here or in #2070. Having many test cases is probably the best way forward.

wsnyder, 2020-06-20:
Yes, we should try to make test_regress cases for the categories of WIDTH mis-warnings then clean them up (or less likely decide not to fix if appropriate). Either add to #2070, or unique bugs.

gezalore, 2020-06-20:
FWIW the "        wb_valid  <= {wb_valid[1],rvtop.swerv.dec.dec_i0_wen_r};" is possibly malformed according to IEEE 1800 7.4.3/7.6 if wb_valid is indeed an upacked array, so a warning there seems fair, though the message could be more helpful.

wallento, 2020-06-20:
@wsnyder right, I can get into it. I suspect that they can be categorized as there is some regularity (without looking deeper into them). I will look into it in 2-3 weeks if noone else picks it up.

wallento, 2020-06-20:
@gezalore is right I think. When I try to replicate the second one I get those warnings:
```
%Error-UNSUPPORTED: t/t_issue-2433.v:9:17: Unsupported: Replication to form 'logic$[0:1]' data type
                                         : ... In instance t
    9 |     wb_valid <= {wb_valid[1], 1'b1};
      |                 ^
%Warning-WIDTH: t/t_issue-2433.v:9:14: Operator ASSIGNDLY expects 1 bits on the Assign RHS, but Assign RHS's REPLICATE generates 2 bits.
                                     : ... In instance t
    9 |     wb_valid <= {wb_valid[1], 1'b1};
      |              ^~
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
```
@agrobman you don't get the error, but only the warning?

wallento, 2020-06-20:
7.6 confuses me, but 10.9.1 seems to be exactly about the issue here. The fix would be:
```diff
-wb_valid <= {wb_valid[1], rvtop.swerv.dec.dec_i0_wen_r};
+wb_valid <= '{wb_valid[1], rvtop.swerv.dec.dec_i0_wen_r};
```
The error from above could be augmented with pointing that out maybe, @wsnyder?
Also, @agrobman can you please try to generate the issues from public code, I cannot find either of these in the EL2 code base..

agrobman, 2020-06-20:
This case is the least problem for us: it’s TB code and pretty limited in usage. Yes, it causes warnings, but simulates OK with all simulators.
I’m not SV expert, but this is kind of array assignment, allowed by SV. Seems Verilator treats it as vector …

First, solve sized constants/parameters warning problem. Our designers were overwhelmed with them and decision was made to disable WIDTH warnings as bogus.
Especially because the complains were not clear and no other simulator warn the code and we had no time to  wait for the Verilator fixes.

Regarding test examples, you have EL2/EH2 RTLs to derive these tests.

I saw also complains about width differences in expressions like:

Logic [1:0] a;
Logic [4:0] b;
Logic [5:0] c;

Assign c = b + a;

(that ‘a’ is too narrow (?))

There are some other warnings,  something about array index is too narrow (?)

===============================

@gezalore<https://github.com/gezalore> is right I think. When I try to replicate the second one I get those warnings:

%Error-UNSUPPORTED: t/t_issue-2433.v:9:17: Unsupported: Replication to form 'logic$[0:1]' data type

                                         : ... In instance t

    9 |     wb_valid <= {wb_valid[1], 1'b1};

      |                 ^

%Warning-WIDTH: t/t_issue-2433.v:9:14: Operator ASSIGNDLY expects 1 bits on the Assign RHS, but Assign RHS's REPLICATE generates 2 bits.

                                     : ... In instance t

    9 |     wb_valid <= {wb_valid[1], 1'b1};

      |              ^~

                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.

@agrobman<https://github.com/agrobman> you don't get the error, but only the warning?

agrobman, 2020-06-20:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

We can’t filter or not filter warnings – in our simulation setup (tools/Makefile) the verilator just dies if there is any warning, so we just disable these warnings to allow simulations.
We don’t have enough experience with the Verilator - it’s not our tool of choice, we were asked just to allow simulations of our cores with it.

wallento, 2020-06-20:
I think I get the misunderstanding here. Other *simulators* will not give you warnings, Verilator will give you exactly 7 that you can deactivate with `-Wno-UNOPTFLAT`. Why is that number so low? Because Verilator is *simulator* and *linter*. That was also why always wondering why you say other tools don't give warnings. Did you mean *linters* or *simulators*? *Simulators* per se are much more forgiving.
There are roughly 4 knobs for tuning how much of a Linter you want in Verilator:
- `-Wno-lint` will just turn the linter part of, leaving you with `-Wno-UNOPTFLAT` if you can't/dont' wannt to fix the loops.
- Once linter is on you can disable entire rules with `-Wno-WIDTH` etc.
- You can also use the inline comment style pragmas `/* verilator lint_off WIDTH */` and `/* verilator lint_on WIDTH */` aroung code blocks.
- The rather recent waiver system to filter individual warnings (I don't really understand your comment why you cannout use it)

So, summarizing, I think we should change the issue title to "Make sure documentation is clear about that Verilator is simulator and linter" (maybe relates to #2439), if @wsnyder agrees that this is the core of this discussion.
Beyond that, I believe this can be closed then. I have the two examples from above covered: one seems indeed a bit too pedantic and I am opening a matching issue for that class of warnings, the other seems to me as not in line with IEEE 1800-2017 10.9.1., but I am wondering why it only gave a warning and not an error.

agrobman, 2020-06-20:
@wallento,

What is "7" warnings you're talking about? You said earlier in PR that you are getting more than 500 (?).

EL2 simulations we are providing with RTL don't generate any warnings, ( Verilator runs with -Wno-WIDTH) because without this switch verilator just dies. (at least versions 4.020 and 4.030 we have installed) and runs correctly.
Are you saying that 4.036 doesn't complain about width mismatched with EL2, except this TB place?

Another thing,  we know nothing about "recent waiver system". Did 4.020 have it? We used 4.020 when released swerv cores!

wallento, 2020-06-20:
As I wrote, it gets down to 7 earnings if you turn linting off.

agrobman, 2020-06-20:
I have 0 with -Wno-WIDTH ..

wallento, 2020-06-20:
Yes, I mean a more recent version than you use.

wallento, 2020-06-20:
> I have 0 with -Wno-WIDTH ..

No, you have 7 UNOPT-FLAT

wallento, 2020-06-20:
Plus it should give you many others, VARHIDDEN etc.

agrobman, 2020-06-20:
look in the tools/Makefile - we use both -Wno...
Let us know what do you get when run
 
make -f $RV_ROOT/tools/Makefile
with 4.020 and the latest

I told you, we don't use verilator as development tool. I can't rape IT to follow every verilator release and re-qualify our design with it.. We don't change tools  version in the middle of project unless there is a problem we can't bypass.

Last time it took 2 weeks to switch from 4.020 to 4.030  because of various issues with linux and other tools ...

agrobman, 2020-06-20:
BTW, Wilson promised to fix this  UNOPT-FLAT problem a few months ago.
We even had a script to replace this code and set up a LEC flow with 1st Swerv to work around this problem. (wasted a lot of time to please verilator :( )
where is the fix?

wallento, 2020-06-20:
As I said before: -Wno-lint will still leave you with UNOPT-FLAT. Do you understand what I mean?

Your complaint seemed to be that lint is too pedantic or bogus. I said I will open individual issues for the WIDTH messages if I think Verilator can be improved for them.

You say you actually don't want lint for your simulation and don't want to update Verilator. So everything is fine, it's safe to close the issue then, please do if you agree.

agrobman, 2020-06-20:
I don't understand what do you want.

You started all this complaining that you got a lot of warnings and we don't run lint.
I explained what we don't see any warnings and why.
I didn't say if I want to use verilator lint or not. I knew nothing about verilator lint at time of our release. I know that commercial simulators issue errors and warnings.
Our check in policy is "no simulator warnings". And we followed it.

I guess -Wno-WIDTH is less masking lint problems than -Wno-lint, so we used this one.
BTW, we consulted with Winston at release time how to suppress these warnings and we followed his advice.

You can't complain that we used older tool version 3 months ago, than one you use now and we even don't know exists.

How do you want us to handle your PR? with what verilator version? Is the latest clean from "bogus" width warnings? If not, when will we get "clean" one?

wallento, 2020-06-20:
> I don't understand what do you want.
>
> You started all this complaining that you got a lot of warnings and we don't run lint.
> I explained what we don't see any warnings and why.
> I didn't say if I [...]

This belongs in https://github.com/chipsalliance/Cores-SweRV-EL2/pull/4
I think its safe to close this issue now.

wsnyder, 2020-06-20:
I think the action item here is for the Verilator devs to look through all swerv width warnings (by removing -Wno-WIDTH) and see what are too verbose/false then fix them.

There is of course community value in having swerv Verilator lint clean including for width, but until its cleaner I see no problem with swerv & others using -Wno-WIDTH.

agrobman, 2020-06-20:
@wsnyder , 

Do you have any plans  to fix UNOPT warning in Swerv ?

wsnyder, 2020-06-20:
I sent one fix in already, for the others there seemed no performance change, so no.

agrobman, 2020-06-20:
Is there way to have warnings and continue to build/simulate with verilator?

gezalore, 2020-06-20:
-Wno-fatal

agrobman, 2020-06-20:
Does it effect errors or only warnings? The name isn’t intuitive …

wallento, 2020-06-20:
Warnings only. You cannot proceed with errors.

gezalore, 2020-06-20:
From the manual:
https://github.com/verilator/verilator/blob/23e4a0f7845334a24a6c083fa52771163905cc00/bin/verilator#L1668-L1671

wsnyder, 2020-06-21:
IMO I do not think swerv or other projects should "ship" with Wno-fatal. Much better to disable by line or even whole categories so the output is clean of warnings.

agrobman, 2020-06-23:
Any updates on this ticket? I got verilator 4.037 installed, seems majority of WIDTH warnings are still incorrect. The tool compares constants/parameter width with other variables width in expressions. :
```
%Warning-WIDTH: design/ifu/elx2_ifu_aln_ctl.sv:196:61: Operator ADD expects 32 or 6 bits on the RHS, but RHS's SEL generates 4 bits.
                                                                                                                    : ... In instance tb_top.rvtop.swerv.ifu.aln
  196 |    localparam                                     MHI   = 46+pt.BHT_GHR_SIZE;
      |                                                             ^
```
pt.BHT_GHR_SIZE is 4 bit wide parameter...

or
```
%Warning-WIDTH: /design/ifu/elx2_ifu_ic_mem.sv:154:47: Bit extraction of var[12:3] requires 4 bit index, not 2 bits.
                                                                                                                   : ... In instance tb_top.rvtop.mem.icache.icm.ic_data_inst
  154 |        ic_debug_sel_sb[i]     = (ic_debug_addr[pt.ICACHE_BANK_HI : pt.ICACHE_BANK_LO] == i );
      |                                               ^

```

wallento, 2020-06-23:
No updates, but we are looking into it. For the first see #2441, this is a pedantic error that will in future probably be in the style-lint, not in the "normal" lint. The style-lint friendly way is:
```diff
-localparam                                     MHI   = 46+pt.BHT_GHR_SIZE;
+localparam                                     MHI   = 46+32'(pt.BHT_GHR_SIZE);
```
Another related issue is #2442 
They will take some time to be properly tested, don't expect them before 4.040.

agrobman, 2020-06-23:
BTW, why this one need to be warned? - this size conversion  makes code less readable. Once a constant fits a variable in LHS  why would  it's size matter?

Why should I write:

bit kuku = 1'b0;
and not
bit kuku = 0;

?
here is where size conversion doesn't work:

```
%Warning-WIDTH: design/ifu/elx2_ifu_bp_ctl.sv:679:52: Operator DIV expects 32 bits on the LHS, but LHS's SEL generates 11 bits.
                                                                                                                   : ... In instance tb_top.rvtop.swerv.ifu.bp
  679 |      for (genvar k=0 ; k < 32'((pt.BHT_ARRAY_DEPTH)/NUM_BHT_LOOP) ; k++) begin : BHT_CLK_GROUP
      |                                                    ^
```

wallento, 2020-06-23:
You can find the argument here: https://github.com/verilator/verilator/issues/2441#issuecomment-647049429
In the other example the conversion is on the DIV operator, you are casting the expression.

gezalore, 2020-06-24:
Regarding your latest. The problem is with types wider than 32 bits (Verilog's semantics re unsized constants are a bit murky when in context wider than 32 bits), so it does not matter so much for `bit x = 0`, but it can cause problems in cases like `god_knows_how_wide_type_defined_somewhere_far_far_away x = -3;`.

I think historically some tools had issues with unsized values as well. I personally have been asked by multiple leading technology companies to not use unsized constants anywhere in the design, so this is really a matter of personal/organisational style so this warning has value. The user friendly solution is to make width warnings more configurable, so you can select at a more fine grained level which ones you want vs not, which is what we will be moving towards.

wsnyder, 2022-11-20:
Closing due to age - reading though this was mostly a discussion thread with some items, fixed, and if any specific WIDTH enhancements are still needed we should file specific tickets.

closed 2022-11-20 by wsnyder