verilator/verilator #2434: Simulation performance Intel Core i7 vs ThreadRipper or MacOS/clang vs Ubuntu/g++

ayewinoung, 2020-06-19:
Hi 
I'm very puzzled by performance "differences" between very modest 2012 Core i7 4Core/8Threads with 16GB RAM MacOS/clang MacBook Pro vs ThreadRipper 12Cores/24threads with 128GB RAM Ubuntu/g++, they both have SSDs.

I have fairly complex RTL/SystemC/C++/C simulation, where the modest MacBookPro get ~ 71 kHz simulation speed with "--threads 8", but ThreadRipper setup is stucked at ~ 34 kHz, regardless of whether "--threads 8" or "--threads 24" in compiler settings.  Both are compiled with exact same "Makefile", except for "--threads".

Both using and compiled for each host.  I have double check nothing else is running in Ubuntu setup.
a. Systemc-2.3.2
b. verilator 4.034

Any thoughts on where the bottle neck for ThreadRipper/Ubuntu/g++ setup could be ?  And pointers on how to improve, "supposedly" better setup to do better?

Appreciate the help in advance.

gezalore, 2020-06-19:
AMD ThreadRipper CPUs have highly non-uniform cache memory architectures, which will cause inefficiencies with the verialted models fine grained multi-threading if the threads are scheduled to physical cores that are far apart (and hence cache maintenance is more expensive). I would start by using numactl to ensure you scheduled all your 8 threads onto the same CCX (see also the MULTITHREADING section in the manual which describes some of this problem). IIRC a CCX should have 4 physical cores/8 physical threads so your --threads 8 build has a chance of having useful performance gains, but in general once you start having to cross to a different CCX I would expect less benefit from multithreading (again IIRC on current ThreadRippers crossing to a different CCX requires going through the IO die, even if it's on the same CCD). I would also recommend the talk by @wsnyder about Verilator multithreading: https://www.youtube.com/watch?v=7nVHUoYja6w. Around 15:15 there is a datapoint on the effect of the "cross socket penalty", as you can see it can be severe. Although the system that was benchmarked in the video sounds like a dual socket system, but architecturally a ThreadRipper is closer to that than a single multicore device like the Core i7 you mention. Full disclosure: I haven't actually tried to run Verilator on a ThreadRipper, that is just what I would expect based on my understanding of the architecture. Also note that GCC vs Clang can make a big difference, so for a more fair comparison I would recommend using Clang on the Ubuntu system as well and if possible the same version of clang (at least so long as you want to compare performance numbers to track this down).

wsnyder, 2020-06-29:
@gezalore well described the impacts here.  I'm assuming this was sufficient for you so will close, but if you have other questions feel free to post - also if you get any data comparing the systems as we'd find it interesting.

closed 2020-06-29 by wsnyder