verilator/verilator #2435: make SIM=verilator in COCOTB error

Divyanirankari, 2020-06-19:
```
make results.xml
make[1]: Entering directory '/home/divya/fbox-verif/mk_divider_dp_instance'
mkdir -p sim_build
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module mk_divider_dp_instance --vpi --public-flat-rw --prefix Vtop -o mk_divider_dp_instance -LDFLAGS "-Wl,-rpath,/home/divya/cocotb/cocotb/libs -L/home/divya/cocotb/cocotb/libs -lcocotbvpi_verilator -lgpi -lcocotb -lgpilog -lcocotbutils"  /home/divya/fbox-verif/mk_divider_dp_instance/mk_divider_dp_instance/mk_divider_dp_instance.v /home/divya/cocotb/cocotb/share/lib/verilator/verilator.cpp
CPPFLAGS="-DVL_TIME_PRECISION_STR=1ps -std=c++11" make -C sim_build -f Vtop.mk
make[2]: Entering directory '/home/divya/fbox-verif/mk_divider_dp_instance/sim_build'
g++  -DVL_TIME_PRECISION_STR=1ps -std=c++11 -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -Os -c -o verilator.o /home/divya/cocotb/cocotb/share/lib/verilator/verilator.cpp
g++  -DVL_TIME_PRECISION_STR=1ps -std=c++11 -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -Os -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
In file included from /usr/local/share/verilator/include/verilated.cpp:25:0:
/usr/local/share/verilator/include/verilated_imp.h: In static member function ‘static IData VerilatedImp::fdNew(const char*, const char*)’:
/usr/local/share/verilator/include/verilated_imp.h:490:78: error: no matching function for call to ‘max(long unsigned int, std::vector<_IO_FILE*>::size_type)’
             const size_t start = std::max(31ul + 1ul + 3ul, s_s.m_fdps.size());
                                                                              ^
In file included from /usr/include/c++/5/deque:60:0,
                 from /usr/local/share/verilator/include/verilated_heavy.h:29,
                 from /usr/local/share/verilator/include/verilated_imp.h:29,
                 from /usr/local/share/verilator/include/verilated.cpp:25:
/usr/include/c++/5/bits/stl_algobase.h:219:5: note: candidate: template<class _Tp> const _Tp& std::max(const _Tp&, const _Tp&)
     max(const _Tp& __a, const _Tp& __b)
     ^
/usr/include/c++/5/bits/stl_algobase.h:219:5: note:   template argument deduction/substitution failed:
In file included from /usr/local/share/verilator/include/verilated.cpp:25:0:
/usr/local/share/verilator/include/verilated_imp.h:490:78: note:   deduced conflicting types for parameter ‘const _Tp’ (‘long unsigned int’ and ‘std::vector<_IO_FILE*>::size_type {aka unsigned int}’)
             const size_t start = std::max(31ul + 1ul + 3ul, s_s.m_fdps.size());
                                                                              ^
In file included from /usr/include/c++/5/deque:60:0,
                 from /usr/local/share/verilator/include/verilated_heavy.h:29,
                 from /usr/local/share/verilator/include/verilated_imp.h:29,
                 from /usr/local/share/verilator/include/verilated.cpp:25:
/usr/include/c++/5/bits/stl_algobase.h:265:5: note: candidate: template<class _Tp, class _Compare> const _Tp& std::max(const _Tp&, const _Tp&, _Compare)
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
/usr/include/c++/5/bits/stl_algobase.h:265:5: note:   template argument deduction/substitution failed:
In file included from /usr/local/share/verilator/include/verilated.cpp:25:0:
/usr/local/share/verilator/include/verilated_imp.h:490:78: note:   deduced conflicting types for parameter ‘const _Tp’ (‘long unsigned int’ and ‘std::vector<_IO_FILE*>::size_type {aka unsigned int}’)
             const size_t start = std::max(31ul + 1ul + 3ul, s_s.m_fdps.size());
                                                                              ^
/usr/local/share/verilator/include/verilated.mk:220: recipe for target 'verilated.o' failed
make[2]: *** [verilated.o] Error 1
make[2]: Leaving directory '/home/divya/fbox-verif/mk_divider_dp_instance/sim_build'
/home/divya/cocotb/cocotb/share/makefiles/simulators/Makefile.verilator:46: recipe for target 'sim_build/mk_divider_dp_instance' failed
make[1]: *** [sim_build/mk_divider_dp_instance] Error 2
make[1]: Leaving directory '/home/divya/fbox-verif/mk_divider_dp_instance'
/home/divya/cocotb/cocotb/share/makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2

```

Divyanirankari, 2020-06-19:
while installing verilator i am gettting error below:
```
../../include/verilated_unordered_set_map.h: In instantiation of ‘vl_unordered_set<T_Key, T_Hash, T_Equal>::const_iterator vl_unordered_set<T_Key, T_Hash, T_Equal>::end() const [with T_Key = std::pair<AstSenTree*, AstActive*>; T_Hash = vl_unordered_map<AstSenTree*, AstActive*>::KeyHash; T_Equal = vl_unordered_map<AstSenTree*, AstActive*>::KeyEqual; vl_unordered_set<T_Key, T_Hash, T_Equal>::const_iterator = vl_unordered_set<std::pair<AstSenTree*, AstActive*>, vl_unordered_map<AstSenTree*, AstActive*>::KeyHash, vl_unordered_map<AstSenTree*, AstActive*>::KeyEqual>::iterator]’:
../../include/verilated_unordered_set_map.h:485:51:   required from ‘vl_unordered_map<T_Key, T_Value, T_Hash, T_Equal>::const_iterator vl_unordered_map<T_Key, T_Value, T_Hash, T_Equal>::end() const [with T_Key = AstSenTree*; T_Value = AstActive*; T_Hash = vl_hash<AstSenTree*>; T_Equal = vl_equal_to<AstSenTree*>; vl_unordered_map<T_Key, T_Value, T_Hash, T_Equal>::const_iterator = vl_unordered_set<std::pair<AstSenTree*, AstActive*>, vl_unordered_map<AstSenTree*, AstActive*>::KeyHash, vl_unordered_map<AstSenTree*, AstActive*>::KeyEqual>::iterator]’
../V3Active.cpp:122:13:   required from here
../../include/verilated_unordered_set_map.h:260:73: warning: large integer implicitly truncated to unsigned type [-Woverflow]
g++ -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG 	 -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3ActiveTop.cpp
In file included from ../V3Ast.h:33:0,
                 from ../V3ActiveTop.h:28,
                 from ../V3ActiveTop.cpp:34:
```

wsnyder, 2020-06-19:
What OS version?

Did you try git master?

Perhaps you could attempt a pulk request to fix this? Looks lije some signed/unsigned type mismatches.

Divyanirankari, 2020-06-19:
@wsnyder  os ubuntu 16.04   32 bit 
ran with master failed to install

Divyanirankari, 2020-06-19:
while doing make after ./configure getting signed/unsigned type error

wsnyder, 2020-06-19:
We nightly build and maintain 64bit builds, can you switch? Note many designs will far exceed 32b limits.

Otherwise if you could provide pulls to fix the compiler issues it would be appreciated.

Divyanirankari, 2020-06-19:
my sytsem is 32bit processor

gezalore, 2020-06-19:
This seems like a duplicate of #2421, and likely fixed by #2426

wallento, 2020-06-20:
@Divyanirankari can you please check out #2426 and check if it solves your issue?
These are the steps:
```bash
git fetch origin pull/2426/head:pr-2426
git checkout pr-2426 
```
And then make distclean and start over again. Thanks!

Divyanirankari, 2020-06-20:
After checkout i need to configure and build again?

wallento, 2020-06-20:
Yes, essentially start from `make distclean` and `autoconf` to be really sure there are artifacts:
```bash
make distclean
autoconf
./configure
make -j `nproc`
make
make test
```

wallento, 2020-06-20:
This should solve the build error, if not I suspect it won't fix the second issue either

Divyanirankari, 2020-06-20:
Okay thanks

Divyanirankari, 2020-06-20:
Verilator build successfully,
while running cocotb/examples/adder/tests getting verilator error:
```
(py36) divya@divya:~/cocotb/examples/adder/tests$ make SIM=verilator
make results.xml
make[1]: Entering directory '/home/divya/cocotb/examples/adder/tests'
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module adder --vpi --public-flat-rw --prefix Vtop -o adder -LDFLAGS "-Wl,-rpath,/home/divya/cocotb/cocotb/libs -L/home/divya/cocotb/cocotb/libs -lcocotbvpi_verilator -lgpi -lcocotb -lgpilog -lcocotbutils"  /home/divya/cocotb/examples/adder/tests/../hdl/adder.v /home/divya/cocotb/cocotb/share/lib/verilator/verilator.cpp
%Error: Invalid Option: --public-flat-rw
%Error: Command Failed /opt/verilator-4/bin/verilator_bin -cc --exe -Mdir sim_build '-DCOCOTB_SIM=1' --top-module adder --vpi --public-flat-rw --prefix Vtop -o adder -LDFLAGS '-Wl,-rpath,/home/divya/cocotb/cocotb/libs -L/home/divya/cocotb/cocotb/libs -lcocotbvpi_verilator -lgpi -lcocotb -lgpilog -lcocotbutils' /home/divya/cocotb/examples/adder/tests/../hdl/adder.v /home/divya/cocotb/cocotb/share/lib/verilator/verilator.cpp
/home/divya/cocotb/cocotb/share/makefiles/simulators/Makefile.verilator:42: recipe for target 'sim_build/Vtop.mk' failed
make[1]: *** [sim_build/Vtop.mk] Error 10
make[1]: Leaving directory '/home/divya/cocotb/examples/adder/tests'
/home/divya/cocotb/cocotb/share/makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2
```

gezalore, 2020-06-20:
What does "/opt/verilator-4/bin/verilator_bin --version" say?

Divyanirankari, 2020-06-20:
Verilator 3.878 2015-11-01 rev UNKNOWN_REV

Divyanirankari, 2020-06-20:
i removed this from PATH now its working.

wsnyder, 2020-06-28:
Looks like this was fixed and is now working for you, thanks for taking the time to report it.

closed 2020-06-28 by wsnyder