verilator/verilator #2437: Documentation issues and observations

gezalore, 2020-06-19:
This is the complementary issue to #2436 with a list of what seems like inaccuracies
in the documentation that I am not comfortable resolving myself because either:
- I don't know what the intended behaviour is or 
- I don't understand the behaviour of Verilator well enough or
- Intention of the text is unclear [to me]
Of course none of this is urgent, and please feel free to ignore comments where I'm just being thick. I numbered them for easy reference, but otherwise they are in text order. Also apologies for the 

One of the things that causes me confusion seems to stem from the fact that in the early days verilator didn't do aggressive module inlining like it does now so I think the user was just able to vTOP->instance->scope->subinstance->subscope->signal  to access signals. The examples use to elude to this, but in light of the module inlining peformed by default I adjusted some of this in #2436, but I wonder if we should have an explicity policy (written down) that explains what the user should and should not assume about the interface of the resulting model (see e.g. the 2nd half of point 10 below).

1. This paragraph is false. Make rules for user files and the executable are only
included iff `--exe` is used:
https://github.com/verilator/verilator/blob/8006b5479c6716a3c7edf1d1478b5e672c797b6d/bin/verilator#L454-L458

2. This last sentence on the language options seem misleading. The only thing the
language seems to influence is the keyword set, some preproc default macros,
and one warning and one error (look for usages of FileLine::language()), but
not any semantics.
https://github.com/verilator/verilator/blob/8006b5479c6716a3c7edf1d1478b5e672c797b6d/bin/verilator#L649-L651

3. Seems to me that `--compiler` is just an alias for specific values of
`--comp-limit-blocks`, `--comp-limit-members` and `--comp-limit-parens`, all
of which are workarounds for compiler issues, but no compiler specific 'tuning'
happens. For a long time I was under the impression this can help performance
based on the documentation, but changing this should have no performance impact
unless one of the workarounds is hit (in which case it might slow things down).
https://github.com/verilator/verilator/blob/8006b5479c6716a3c7edf1d1478b5e672c797b6d/bin/verilator#L612-L614

4. Might need updating in light of recent changes which measure coverage for all
statements:
https://github.com/verilator/verilator/blob/8006b5479c6716a3c7edf1d1478b5e672c797b6d/bin/verilator#L649-L657

5 Is --pipe-filter still experimental? It was added in 2010.
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L1202-L1204

6. --public-flat-rw suggests it marks ports and variables public:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L1323-L1331
But the implementation suggests it only marks wires, though I'm not sure if
this is equivalent or not as I am not familiar with LinkParse:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/src/V3LinkParse.cpp#L197-L204
It is also confusing that the public_flat_rw meta-comment takes an edge-list
but this option does not mention that at all.
The short list also talks of variables:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L364

7. --prof-threads is not a "simulation runtime" option but a "verilation time"
option. Not sure how to use this properly.
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L1855-L1875

8. Wonder if this section is still relevant in light of current Cadence tools.
Plus with my non-hardware-engineer hat on, it feels a little bit like it's
trying to say "you can compile C++ with anything that can compile C++":
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L2809-L2827

9. Is multithreading still considered experimental?
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L2830-L2832

10. I don't think there is now any tracing difference with or without module
inlining. I also don't understand what "modularity" means here. Also I'm not
sure the part about __DOT__ names is particularly helpful. Internal names are
internal (unless marked public), so should the user care at all? This text
sounds like it's from earlier days and might need updating.
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3047-L3057

Same appears in:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3394-L3404

11. Very confused by what "This is useful especially at the top level module to
reduce the size of the interface class, to aid compile time at a small
performance loss." is trying to say:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3073-L3081

Same appears in:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3473-L3481

12. Also very confused by this sentence. Is it trying to say "if you want
a parametrized width port to always be sc_bv" ?
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3101-L3103

Same appears in:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3600-L3602

13. None of these are documented in CONFIGURATION FILES, though all of them claim
to have an equivalent
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3510-L3588

14. Again confusion about ports/variables/wires with public_flat_rw.
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3568-L3574

15. How will the known scopes be printed?
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3722-L3723

16. Do we know what cases cause problems for scheduling (if this is still
applicable)? This is a bit scary these days as designs often have ICGs, but
I never noticed a problem.
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3817-L3824

17. I think #2223 added a bit more, though I am not sure of the exact rule:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3895-L3899

18. Might need updating in light of #2197:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L3982-L3986

19. GENCLK is never issued (V3Order has it commented out), also the note about this
causing simulation mismatch is suspect:
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L4350-L4361

20. Is this paragraph still relevant? (it's from the 2006 root commit)
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L5004-L5006

21. Should we still mention Dinotrace?
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L5156-L5161

22. Bit confused by this, is this trying to say "use the DPI, *or* a /*verilator
public*/ task/function", similar to the next question about accessing signals?
https://github.com/verilator/verilator/blob/fa828ac9badc0531a28ae888a567d3237391fe9f/bin/verilator#L5306-L5311

wsnyder, 2020-06-20:
Agree all if these need fixing, I'll hit them all before the next release.

wsnyder, 2020-06-28:
Think I fixed most of the items above, only one confused me:

```
if (v3Global.opt.publicFlatRW()) { 
     switch (nodep->varType()) { 
     case AstVarType::VAR: 
     case AstVarType::PORT: 
     case AstVarType::WIRE: nodep->sigUserRWPublic(true); break; 
```

Note the VAR/PORT are fall through (I'll add a comment) so this indeed is marking variables, ports or wires so the documentation seemed correct.  Or, let me know if I missed your point.

closed 2020-06-28 by wsnyder