verilator/verilator #2441: WIDTH: propagation through arithmetic

wallento, 2020-06-20:
This is the first one towards #2433 
I have seen this quite often and it is kind of common with parameters:
```
module t;

  localparam logic[3:0] A = 4'h0;
  localparam B = 46 + A;
  localparam logic[12:0] C = 46 + A;

endmodule
```
Leads to:
```
%Warning-WIDTH: t/t_issue-2433.v:4:21: Operator ADD expects 32 or 6 bits on the RHS, but RHS's VARREF 'A' generates 4 bits.
                                     : ... In instance t
    4 |   localparam B = 46 + A;
      |                     ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: t/t_issue-2433.v:5:33: Operator ADD expects 32 or 13 bits on the RHS, but RHS's VARREF 'A' generates 4 bits.
                                     : ... In instance t
    5 |   localparam logic[12:0] C = 46 + A;
      |                                 ^
```
For line 4:
 - (1800-2017 11.6.1) size of the add is max(size(46)=6, size(A)=4)=6
 - (1800-2017 11.8.2) propagate back size 6 to 46 and A.
 - (1800-2017 11.8.3) RHS=6, LHS/context=32 -> RHS=32

I looked at `V3Width` before when I tried to solve #2070. It seems kind of in the same area (up- and down-propagating). It uses the min/max sizes (thats where the 32 comes from) and I believe there is an issue when back-propagating from assignment to the ADD.
For line 5 it should be the same, but only with context=13->RHS=13.
Either way, my understanding from the spec is that both statements are actually okay, or am I missing something?
This removes the warning by the way:
```
module t;

  localparam logic[3:0] A = 4'h0;
  localparam B = 46 + 32'(A);
  localparam logic[12:0] C = 46 + 13'(A);

endmodule
```

wallento, 2020-06-20:
By the way, not sure if the title is actually the problem, just an impression I had before.

wallento, 2020-06-20:
This is the tree:
```
Verilator Tree Dump (format 0x3900) from <e85> to <e95>
     NETLIST 0x55bc30bcbf80 <e1> {a0aa}  $root [1ps/1ps]
    1: MODULE 0x55bc30be26e0 <e85#> {d1ai}  t  L0 [1ps]
    1:2: VAR 0x55bc30be3e80 <e24> {d3az} @dt=0@  A [VSTATIC]  LPARAM
    1:2:1: BASICDTYPE 0x55bc30be39a0 <e25> {d3ao} @dt=this@(nw1)  logic kwd=logic
    1:2:1:1: RANGE 0x55bc30be3a80 <e18> {d3at}
    1:2:1:1:2: CONST 0x55bc30be3b40 <e16> {d3au} @dt=0x55bc30be2dc0@(G/swu32/2)  ?32?sh3
    1:2:1:1:3: CONST 0x55bc30be3cd0 <e17> {d3aw} @dt=0x55bc30be3230@(G/swu32/1)  ?32?sh0
    1:2:3: CONST 0x55bc30be3650 <e26> {d3bd} @dt=0x55bc30be3800@(G/w4)  4'h0
    1:2: VAR 0x55bc30be4ac0 <e47> {d4ao} @dt=0@  B [VSTATIC]  LPARAM
    1:2:1: BASICDTYPE 0x55bc30be49e0 <e45> {d4ao} @dt=this@(nw0)  LOGIC_IMPLICIT kwd=LOGIC_IMPLICIT
    1:2:3: ADD 0x55bc30be4920 <e46> {d4av} @dt=0x55bc30be4600@(G/swu32/6)
    1:2:3:1: CONST 0x55bc30be4450 <e39> {d4as} @dt=0x55bc30be4600@(G/swu32/6)  ?32?sh2e
    1:2:3:2: VARREF 0x55bc30bdf7f0 <e88#> {d4ax} @dt=0@  A [RV] <- VAR 0x55bc30be3e80 <e24> {d3az} @dt=0@  A [VSTATIC]  LPARAM
    1:2: VAR 0x55bc30c31220 <e79> {d5ba} @dt=0@  C [VSTATIC]  LPARAM
    1:2:1: BASICDTYPE 0x55bc30c30d40 <e77> {d5ao} @dt=this@(nw1)  logic kwd=logic
    1:2:1:1: RANGE 0x55bc30c30e20 <e64> {d5at}
    1:2:1:1:2: CONST 0x55bc30c30ee0 <e62> {d5au} @dt=0x55bc30c2ffc0@(G/swu32/4)  ?32?shc
    1:2:1:1:3: CONST 0x55bc30c31070 <e63> {d5ax} @dt=0x55bc30be3230@(G/swu32/1)  ?32?sh0
    1:2:3: ADD 0x55bc30c30c80 <e78> {d5bh} @dt=0x55bc30be4600@(G/swu32/6)
    1:2:3:1: CONST 0x55bc30c30800 <e73> {d5be} @dt=0x55bc30be4600@(G/swu32/6)  ?32?sh2e
    1:2:3:2: VARREF 0x55bc30bdf910 <e91#> {d5bj} @dt=0@  A [RV] <- VAR 0x55bc30be3e80 <e24> {d3az} @dt=0@  A [VSTATIC]  LPARAM
    3: TYPETABLE 0x55bc30bcc620 <e2> {a0aa}
		detailed  ->  BASICDTYPE 0x55bc30be3800 <e22> {d3bd} @dt=this@(G/w4)  logic [GENERIC] kwd=logic range=[3:0]
		detailed  ->  BASICDTYPE 0x55bc30be3230 <e13> {d3aw} @dt=this@(G/swu32/1)  logic [GENERIC] kwd=logic range=[31:0]
		detailed  ->  BASICDTYPE 0x55bc30be2dc0 <e8> {d3au} @dt=this@(G/swu32/2)  logic [GENERIC] kwd=logic range=[31:0]
		detailed  ->  BASICDTYPE 0x55bc30c2ffc0 <e53> {d5au} @dt=this@(G/swu32/4)  logic [GENERIC] kwd=logic range=[31:0]
		detailed  ->  BASICDTYPE 0x55bc30be4600 <e35> {d4as} @dt=this@(G/swu32/6)  logic [GENERIC] kwd=logic range=[31:0]
    3:1: BASICDTYPE 0x55bc30be2dc0 <e8> {d3au} @dt=this@(G/swu32/2)  logic [GENERIC] kwd=logic range=[31:0]
    3:1: BASICDTYPE 0x55bc30be3230 <e13> {d3aw} @dt=this@(G/swu32/1)  logic [GENERIC] kwd=logic range=[31:0]
    3:1: BASICDTYPE 0x55bc30be3800 <e22> {d3bd} @dt=this@(G/w4)  logic [GENERIC] kwd=logic range=[3:0]
    3:1: BASICDTYPE 0x55bc30be4600 <e35> {d4as} @dt=this@(G/swu32/6)  logic [GENERIC] kwd=logic range=[31:0]
    3:1: BASICDTYPE 0x55bc30c2ffc0 <e53> {d5au} @dt=this@(G/swu32/4)  logic [GENERIC] kwd=logic range=[31:0]
```

wallento, 2020-06-20:
Ah, just discovered this is not `V3Width` but `V3Param`, so my very small wisdom is even useless :)

wsnyder, 2020-06-20:
These warnings are correct for the current definition of WIDTH. It is saying that A should be wider to match the other expression side, or a cast used to acknowledge that there are too few bits being added. I like this because it finds missized CSRs and such.

Perhaps what some people are preferring is a stronger warning that there is e.g. truncation going on.  (wire [3:0] = 5'b11111).

wallento, 2020-06-20:
Yeah, that was what I was also thinking, that there could be two classes WIDTH and WIDTH-EXACT or so.

gezalore, 2020-06-20:
On another note, if both sides are known constant (as in the example), do we care, so long as the result fits in the target width [sloppy but works]?

wallento, 2020-06-20:
@wsnyder can you please give a brief example of what it can detect? I thought I had it and wanted to explain to geza and then found I actually didn't..

wallento, 2020-06-20:
By the way, this could be a chance to create content for your suggested stylishsv. It seems like a pedantic thing to make sure for summing up two constants to a third one to ensure both are equally sized, but a good example probably gives people a better understanding of the issues this may lead to.

wallento, 2020-06-20:
```
module t;

  localparam logic[3:0] A = 4'h0;
  localparam B = 46 + 32'(A);
  localparam C = 46 + 6'(A);
  localparam D = 46 + 11'(A);
  localparam E = 10'(46) + 11'(A);
  localparam F = 46 + A;

endmodule
```
Out of those, only F is a WIDTH warning, this seems a bit odd, I would have expected E to also give the warning.

wallento, 2020-06-20:
Okay, I get more confused:
```
  localparam G = 46 + 5'(A);
  localparam H = 46 + 4'(A);
```
I would expect both to fail, but only H fails.

gezalore, 2020-06-20:
FWIW about 85% of width warnings over binary ops in EL2 have at least one compile time constant operand.

wsnyder, 2020-06-21:
Note cases where one operand as 1 larger intentionally didn't warn because this is a common convention to add a carry.

Honestly the WIDTH stuff was optimized ti suit 3 companies I worked at. Ideally we should survey other tools and ideas and try to be similarly strict or not, and probably fracture into several warnings with some probably style only.

wallento, 2020-06-21:
I have added the branch `width-exact` that adds this class. I suspect there will be a handful of checks in `widthCheckSized` for the kind of errors described here. In #2442 it will probably the old (WIDTHEXACT) vs the new (WIDTH) code.

gezalore, 2020-06-21:
Sorry I'm not sure I follow your last comment.

wallento, 2020-06-21:
@gezalore I meant the part that you are relaxing. Keep the old one as the "strict/pendantic/exact" one and have the new too.

wsnyder, 2023-08-30:
Was fixed a while ago and now gives extend warning.

closed 2023-08-30 by wsnyder