verilator/verilator #2446: Scheduling features needed for UVM support

tgorochowik, 2020-06-26:
### Introduction

Our goal is to be able to use UVM with Verilator.

Note that we completely ignore the fact that currently the whole UVM cannot even be parsed by Verilator, let's focus on scheduling here.
The rest is a part of a separate effort.

There are two significant things that need addressing:
* lack of support for the `fork`/`join` statements
* lack of events triggering (`->my_event`) and waiting for them inside tasks (`@ my_event`)

The ideal way to solve this would be jump right into handling event driven simulation and implementing a _full_ stratified scheduler.

We want to discuss other possible ways to implement what is needed.
Maybe it would be possible to make the first implementation a little bit more simplified.
Below are some notes on various approaches that could be taken.

### Execute forked code sequentially

The idea here is that maybe we could execute the following code:

```verilog
fork
  <block A>
  <block B>
  <block C>
join
```

in sequence (i.e. execute A then B then C - not in parallel).

We would probably have to show a warning when doing that as not everything would work as expected with this approach.

Advantages:
* the biggest benefit of this would be that it would be the easiest to implement
* would allow us to move forward with the implementation of stuff missing for UVM

Disadvantages:
* will work with a limited number of cases
* will have to be re-visited sooner or later

This is also how regular blocks (`initial` or block reacting to the same edge) are used.

### Combining `fork`/`join` with `#delay` timing

Assuming https://github.com/verilator/verilator/pull/2409 - or some other implementation goes to master, the approach described above would have to be different.

Each forked block would have to be represented as a separate FSM (similarly to what the code in the branch does for each `initial` block, etc.).
The new FSM would have to be nested in the ones handling the code where the `fork` statement is used.

### Handling event triggers

There are various ways the events can be used.
However the way UVM uses events in most cases is fairly simple, here is an example:

```systemverilog
virtual task wait_trigger ();
  num_waiters++;
  @m_event;
endtask
```

So basically this boils down to blocking the execution until somebody triggers the event with `->m_event`.

The simplest way to implement this would be using a FSM mechanism similar to #2409.
Calling `->` could advance the state in the machine, or maybe just set some boolean flag that would allow the machine to advance itself during the next eval loop.

We also need to support the `.triggered` method for events.
It should be fairly easy to add a special case to handle that.

### Implementation of a proper stratified scheduler

This would probably be the most preferable way as this would be the closest to what should be done according to the IEEE.
There are many regions that could be ignored in the first implementation (namely the PLI-only regions).

All regions:
* Preponed
* Pre-Active
* Active
* Inactive
* Pre-NBA
* NBA
* Post-NBA
* Pre-Observed
* Observed
* Post-Observed
* Reactive
* Re-Inactive
* Pre-Re-NBA
* Re-NBA
* Post-Re-NBA
* Pre-Postpone
* Postponed

PLI specific regions which can be ignored for now:
* Preponed
* Pre-Active
* Pre-NBA
* Post-NBA
* Pre-Observed
* Post-Observed
* Pre-Re-NBA
* Post-Re-NBA
* Pre-Postponed

So we are left with 8 regions:
* Active
* Inactive
* NBA
* Observed
* Reactive
* Re-Inactive
* Re-NBA
* Postponed

Implementing the stratified scheduler would still be the most time consuming approach.
The main issue with that is it would have to modify lots of the features that are already supported as even assignments would have to go through the queue.

We are willing to start our implementation with the simple approaches suggested above.
If we find some blockers or come up with new ideas during our work we will try to keep this updated.

We create this issue here before starting our work so you can share your comments if there is some early feedback or some blockers we did not anticipate or if there is something fundamentally wrong with our approaches.
Please do so!

@wsnyder when we talked about this some time ago you mentioned that this will not be doable without using actual threads for the implementation.
Could you please elaborate on that?

wsnyder, 2020-06-28:
### Introduction

>we completely ignore the fact that currently the whole UVM cannot even be parsed by Verilator,

To be pedantic, in git master and the next planned release (4.038) Verilator can parse all of UVM, but cannot elaborate it (yet).

### Handling event triggers

>Calling `->` could advance the state in the machine, or maybe just set
>some boolean flag that would allow the machine to advance itself during
>the next eval loop.

Currently events are just (references to) booleans, I agree this still will probably be sufficient.

>We also need to support the `.triggered` method for events.
>It should be fairly easy to add a special case to handle that.

Agreed; as simple events are already handled today, this should be straightfoward to add as a relatively contained patch.

### Execute forked code sequentially

>The idea here is that maybe we could [sequentially] execute the following code:

```verilog
fork
  <block A>
  <block B>
  <block C>
join
```

Once there are any delays in A/B/C sequential execution won't work.  Having delays is very necessary to write transactors.

The FSM approach in #2049 would fix that and allow delays. However I don't see how the FSM approach can support join_any/join_none as it cannot know how to break the blocks appropriately.

Since UVM uses join_none in e.g. uvm_sequencer_base::m_wait_for_available_sequence and uvm_reg_backdoor::start_update_thread, so I believe real process-based fork/join support is required for UVM.

### Process

The other complication is supporting "process". UVM requires this.

'''
function void uvm_reg_backdoor::start_update_thread(uvm_object element);
   ...
   this.m_update_thread[element] = process::self();
'''

Once you have to support a reference to a process I don't see how a FSM approach can work.

### Implementation of a proper stratified scheduler

>This would probably be the most preferable way as this would be the closest to what should be done according to the IEEE.
>There are many regions that could be ignored in the first implementation q(namely the PLI-only regions).
>
>Implementing the stratified scheduler would still be the most time consuming approach.
>The main issue with that is it would have to modify lots of the features that are already supported as even assignments would have to go through the que$
>
>We are willing to start our implementation with the simple approaches suggested above.
>If we find some blockers or come up with new ideas during our work we will try to keep this updated.

UVM doesn't use 'program' so theoretically there might be the ability to have some success without redoing the scheduler.  Looking at some UVM test benches I have access to I do see use of "program" and suspect they will not work without the stratification.

I it's probably reasonable to work on the fork/process handling stuff first, as there's no question on that, and it is fairly orthagonal to the scheduling model. Once we have that present we can get an idea of the scheduler importance.

The caution is if we do need the scheduler changes, that will likely be the critical path, so this approach may delay the ultimate date that UVM is working.

For this reason if we have enough people to work on both fork and stratified I think attacking both would be desired, but if it's the same people working on both then fork is the critical one.

wsnyder, 2020-06-29:
Also note UVM uses semaphore and mailbox get() which must block the current process and switch to another process.

martin-lueker, 2020-07-01:
Thanks @wsnyder for your feedback.   So, as you say the process class is necessary to UVM in general.    This does seem like a big lift, but if one was able to implement this it seems that the same mechanism would lead to almost a trivial implementation of fork/join.   Would you agree?

wsnyder, 2020-07-01:
Most of the work is process.  With process fork/join should be easy thought we might want a few simple optimizations (for example empty fork/joins). Some small tweaks to event.  Then the final medium chunk is proper time passing (in process or otherwise).

martin-lueker, 2020-07-01:
Right.  Though really here `process`  is the general abstraction that if you get it right allows you to cleanly implement everything else.

I can imagine fairly simple implementations. For instance, instead of creating a single scheduler queue you have multiple scheduler queues and you just continuously rotate through them, round-robin, tracking the current simtime, and resuming each queue as events come up.   Forks would just create more queues at runtime, and joins would describe the type of blocking event for the parent queue/process.  I'm sure other implementations are out there.  

The devil would be in the details.  For example, I don't have a clear idea how you would associate particular blocks of code to a particular queue.    

Regardless I see your point though, @wsynder. If you eventually want to implement both fork/join and and a stratified scheduler, you would probably want to design the scheduler within the context of a generic process framework.

martin-lueker, 2020-07-01:
Right, thanks for the clarification @wsnyder.  I understand from a syntactic point of view, I was more thinking from an implementation point of view, though I realize that verilator has to solve similar problems for static processes, so maybe that detail won't be so bad.

tgorochowik, 2020-07-02:
@wsnyder so to sum up - in your opinion it would be the best to start with implementing some basic support for processes, and for the time being ignore stratified scheduling?

Could you please give us some pointers on how would you approach the implementation so we are going in the right direction right from the start? We could still start with the simplest approaches I described in the original post, but if you think they will have to be redone anyway after changing the scheduler then we are open to suggestions.

wsnyder, 2020-07-02:
Yes, after this conversation it does seem that processes is the best big task to work on first.

I'll think about implementation.  Perhaps we can leverage some C++ coroutine support...

tgorochowik, 2020-08-27:
Here are some notes from our call which determine the next steps regarding this issue:

- In the end we should end up with a mixed implementation - static scheduling where possible and using threads where we can’t avoid them 
- The static scheduling should use the FSM approach as much as possible
- We should create a visitor that goes through the tree and checks which parts can be represented as a FSM and which parts need to be represented as threads
- The visitor should probably be put after scoping and before ordering
- For now this visitor will only be used to mark parts of the verilog code as unsupported (as the FSM would easily break if we wanted to represent it that way)
- At a later stage, it will serve to mark entry points for creating actual C threads
- In the meantime we should come up with some examples that would show where we can’t avoid threads - we could add them to verilator right away as tests that are expected to fail

CC: @wsnyder @martin-lueker @piotr-binkowski

wsnyder, 2020-12-02:
Did some research on coroutines. There's some good-seeming options:
1. C++20 coroutines.  Seems best option but C++20 is probably too new.  Perhaps we do something else but use coroutines if using C++20.
2. Boost coroutines.   Seems what we need but I'm reluctant to pull in the huge boost libraries.  https://www.boost.org/doc/libs/1_74_0/libs/coroutine/doc/html/index.html
3. Tencent libco.  Reasonable, small, no recent activity, Linux only. https://github.com/Tencent/libco
4. Co. https://github.com/idealvin/co
5. Libcopp. Cross platform, C++20 support. https://github.com/owt5008137/libcopp

Would suggest look at 5th most closely, tests passed out-of-the-box for me.

udif, 2021-04-27:
I've had been researching this in the background since opening #2035.
I've found https://github.com/lewissbaker/cppcoro and have seen it mentioned elsewhere too. It's supposed to be compliant with C++20. It seems to be oriented towards clang, and I'm not sure if/what g++ version is supported.

As this is a new feature, it seems reasonable to me to expect the user to have a C++20 compiler if he wants this feature.
After all, this only affects the compiler used for the runtime, not the compiler used for building Verilator, and even for runtime, C++20 is required only if this feature is used. It does not effect existing users.

BTW, the link for #5 has changed to https://github.com/owent/libcopp (username change). The original link still forwards to the new one, though.

wsnyder, 2022-05-15:
#3363 when merged should fix this.

wsnyder, 2022-06-20:
Closing, as think this discussion is done now that #3363 has an implementation.

closed 2022-06-20 by wsnyder