verilator/verilator #2447: VPI handles to parameters

liza-zoubakina, 2020-06-26:
Hi, I'm currently working on an integration of cocotb and Verilator to run simulations. I am new to verilator and currently learning how it works.
I am working with Verilator 4.037 dev version and latest version of Cocotb, on MacOS. The reason why I chose to work with the dev version is that I heard that support for accessing parameters through VPI was added very recently.

I encountered the following issues where:
- Cocotb is having trouble reading certain parameter values from verilator. For example, my parameters are initialized in the following format in the top level file:
`module example #( parameter EXAMPLE = 1, parameter y = 2 ) ...`
- The simulations, which previously ran on icarus and vcs with success, no longer run as they cite an `AttributeError` where
`AttributeError: top_level_tile contains no object named EXAMPLE`

This error still occurs when I switched to the dev version of Verilator, and so I am confused as to why Verilator still cannot let the parameters be accessed through VPI. 

- As a way to bypass the issue, I declared the parameters as vectors instead. Such as: `wire [31:0] EXAMPLE = 1;`
- However, this led to additional issues where cocotb began reading the value of `EXAMPLE` as zero instead of 1. 
- I used gtkwave to read the dumped waveforms of the simulation, however the wire signals are read as 1. This led me to suspect that cocotb is not properly reading the registers that are initialized in verilator.

- Finally, through backtracking the bug, I suspect that the GPI interface between cocotb and verilator is likely not sending the correct values of the registers. 

I am not sure as to why cocotb still has trouble reading wire signals from verilator. If you have any ideas or pointers, please let me know, as I am not very familiar with the VPI/GPI interfaces between cocotb and a simulator.

elgorwi, 2020-06-26:
Note the parameters need to be explicitly made public to make them visible from VPI.
Can you try following:
```
module example #( 
parameter EXAMPLE /*verilator public_flat_rd*/ = 1, 
parameter y /*verilator public_flat_rd*/ = 2 )
```

liza-zoubakina, 2020-06-29:
Thank you! I have tried it and those explicit public declarations have worked and the values of the parameters are properly initialized and read now.
Apparently cocotb is still pending a merge on accessing parameter VPIs (https://github.com/cocotb/cocotb/pull/1901) in `cocotb/share/lib/vpi/.`.. so I will look out for that as well.

wsnyder, 2020-06-29:
@liza-zoubakina glad to hear it!  I'm presuming we can close this, feel free to reopen or comment.

closed 2020-06-29 by wsnyder