verilator/verilator #2461: Verilator is unable to compile NVidia's DLA

jameshanlon, 2020-07-11:
A change in Verilator in the last year has appeared to withdraw support for parameters in functions. This means that it is no longer able to compile NVidia's DLA core (https://github.com/nvdla/hw).

Steps to reproduce:
```
git clone https://github.com/nvdla/hw.git
cd hw
make
./tools/bin/tmake -build vmod
cd verif/verilator
verilator --lint-only -f verilator.f --timescale-override 1ns
%Error-UNSUPPORTED: ../../outdir/nv_full/vmod/vlibs/HLS_cdp_icvt.v:107:17: Unsupported: Parameters in functions
  107 |       parameter olen = width_z;
      |                 ^~~~
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_cvtin.v:200:1: ... note: In file included from NV_NVDLA_CDP_DP_cvtin.v
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_dp.v:315:1: ... note: In file included from NV_NVDLA_CDP_dp.v
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_cdp.v:327:1: ... note: In file included from NV_NVDLA_cdp.v
                    ../../outdir/nv_full/vmod/nvdla/top/NV_NVDLA_partition_o.v:2123:1: ... note: In file included from NV_NVDLA_partition_o.v
                    ../../outdir/nv_full/vmod/nvdla/top/NV_nvdla.v:1134:1: ... note: In file included from NV_nvdla.v
%Error-UNSUPPORTED: ../../outdir/nv_full/vmod/vlibs/HLS_cdp_icvt.v:108:17: Unsupported: Parameters in functions
  108 |       parameter ilen = width_a+1;
      |                 ^~~~
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_cvtin.v:200:1: ... note: In file included from NV_NVDLA_CDP_DP_cvtin.v
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_dp.v:315:1: ... note: In file included from NV_NVDLA_CDP_dp.v
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_cdp.v:327:1: ... note: In file included from NV_NVDLA_cdp.v
                    ../../outdir/nv_full/vmod/nvdla/top/NV_NVDLA_partition_o.v:2123:1: ... note: In file included from NV_NVDLA_partition_o.v
                    ../../outdir/nv_full/vmod/nvdla/top/NV_nvdla.v:1134:1: ... note: In file included from NV_nvdla.v
%Error-UNSUPPORTED: ../../outdir/nv_full/vmod/vlibs/HLS_cdp_icvt.v:109:17: Unsupported: Parameters in functions
  109 |       parameter len = (ilen >= olen) ? ilen : olen;
      |                 ^~~
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_cvtin.v:200:1: ... note: In file included from NV_NVDLA_CDP_DP_cvtin.v
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_CDP_dp.v:315:1: ... note: In file included from NV_NVDLA_CDP_dp.v
                    ../../outdir/nv_full/vmod/nvdla/cdp/NV_NVDLA_cdp.v:327:1: ... note: In file included from NV_NVDLA_cdp.v
                    ../../outdir/nv_full/vmod/nvdla/top/NV_NVDLA_partition_o.v:2123:1: ... note: In file included from NV_NVDLA_partition_o.v
                    ../../outdir/nv_full/vmod/nvdla/top/NV_nvdla.v:1134:1: ... note: In file included from NV_nvdla.v
...
```
Is this expected? And is there a way of re enabling support for function parameters, or working around this issue?

wsnyder, 2020-07-14:
Parameters in functions were never intentionally supported, but might have worked in (this) limited case by accident.

If you could make a test_regress format test case that checks the parameters work properly (e.g. instantiate the module with the function many different ways), or ideally also investigate removing the rest we can hopefully get this fixed quickly.

wsnyder, 2020-07-15:
The simple parameters supported in past versions should now pass again; the broken cases prompting adding the error now give a more specific error.

I was unable to test NVidia DLA - got "Cannot find file containing module: ../../outdir/nv_full/vmod/nvdla/top/NV_nvdla.v" please confirm if it now works.

jameshanlon, 2020-07-16:
Hi Wilson, thanks for getting on to this so quickly - didn't give me a chance to write a test case!

I have tried building nvdla again with the fix and it works.

To try it out I've checked in the generated RTL:
```
git clone https://github.com/jameshanlon/hw.git
git checkout build
cd hw/verif/verilator
verilator +1800-2012ext+.sv --xml-only --flatten --error-limit 10000 --xml-output out.xml -f verilator.f  --timescale-override 1ns
```

closed 2020-07-16 by jameshanlon

flylcl, 2021-03-04:
Recommend to run simulation by higher version of Verilator