verilator/verilator #2466: _eval() seems evaluate the combo logic firstly?

simdream, 2020-07-14:
From internals.adox (and one presentation years ago):

On other evaluations, the Verilated code detects what input signals have changes. If any are clocks, it calls the appropriate sequential functions (from always @ posedge statements). Interspersed with sequential functions it calls combo functions (from always @*). After this is complete, it detects any changes due to combo loops or internally generated clocks, and if one is found must reevaluate the model again.

This seems like the evaluate flow I agree, but from the generated _eval(), I can see combo logic are evaluated firstly, which is not aligned with the wording above.

Reason asking is we have two evaluate wtice in one click tick. 

Give Verilog exmaple:
module TB;
..
always @(posedge clk)
    sig_dut = ...;

dut dut (.in(sig_dut), ...)

module dut(in, ...);
wire d;
d=func( in, ...);
always @(posedge clk) 
   q<= d;
endmodule

To write C++ TB, within the while loop we have to do:
while(){
   Vdut->in = sig_dut;
   eva();
   ftp->dump(mainTime);  // current simulation time
   mainTime++;
   eval();
   ftp->dum(mainTime); // advanced to the next time
}

If we donot eval() twice, the "d" signal change at the same time as q. I think the function of "q" is still correct, just the "d" and "q" flip at the same clock edge hints race-condition like situation. Any suggest here ?

wsnyder, 2020-07-14:
Verilator attempts to statically schedule the combo code, and so will in many cases evaluate it "before" the flops in order to avoid extra loops inside the eval code.

You don't show where clocks are updated in your C++ code, but most likely you are doing it exactly when you set "in". You should get the results you want if you don't change "in" on the same edge (eval call) as you change the clocks.

I'm presuming this answers your question so am closing but feel free to ask more here, or reopen.

closed 2020-07-14 by wsnyder

simdream, 2020-07-15:
Add where the clock is update, it is right before the second eval().

while(){
   Vdut->in = sig_dut;
   eval();
   ftp->dump(mainTime);  // current simulation time
   mainTime++;

   clk = clk? 0 : 1;
   Vdut->clk = clk;    // clk update...
   eval();
   ftp->dump(mainTime); // advanced to the next time
}

If the first eval() and dump() are removed, I assume the function of design might still be correct. But then the waveform of "d" looks wired. To have two eval()s in one tick time is also time consuming.

wsnyder, 2020-07-15:
It might be there's no race then but the waves is confused as time it's passing. Better to have some time pass there in the middle to avoid trace confusion.