verilator/verilator #2471: how to debug if compile never ends

simdream, 2020-07-17:
We are doing a test case (as attached). By turning on --debug, we can see some messages and when the compile seems run into dead loop. But after that, any other method or message can help debug other than run into C code single step? 

The code might have something wired as it failed in iverilog as well. But it passed vivado xsim.

[test.zip](https://github.com/verilator/verilator/files/4936908/test.zip)

wsnyder, 2020-07-17:
If you run with "--debug" it seems to hang in V3LinkDot, so then use "--debug --gdbbt" and hit Ctrl-C and can see the C code it's hitting, and/or "--debug --debugi-V3LinkDot 9" to trace the statements.

If you could look a bit into it, and also maybe try reducing the test case (e.g. probably don't need so many instantiations), that would help, I'll try to get back to this also in a day or so.

wsnyder, 2020-08-10:
Looked into this. Compiling this will always be slow as there's 25,000 submodules.  Looked for any O(n^2) behavior and found V3Combine goes O(n^2), due to this specific test case really having many potential duplicates.   Some improvement would be desirable there.

wsnyder, 2026-10-18:
While still takes a while V3Combine itself seems ok on master.

closed 2026-10-18 by wsnyder