verilator/verilator #2472: what next if we see one Mtask takes long time in multi threading run?

simdream, 2020-07-17:
Using Ariane RISCV for a "--threads 2" compile and and run. Attached is the profile. It shows Mtask#8 consume most of the time so multi threading compile clearly is not well balanced. But start from here, how can we do anything to balance?

Also my understand is that everytime there is a eval() called from testbench, eval() will start fork for multi threads. How can we read the message that thread#2 is idle for so long simulation time? Shouldn't thread#2 be at least forked every eval() and do little something?

[image]

simdream, 2020-07-17:
grep Vmtask_ * | more

Variane_testharness.cpp:    vlTOPp->__Vm_threadPoolp->workerp(0)->addTask(__Vmtask__7, vlTOPp->__Vm_even_cycle, vlSymsp);
Variane_testharness.cpp:    __Vmtask__8(vlTOPp->__Vm_even_cycle, vlSymsp);
Variane_testharness.cpp:void Variane_testharness::__Vmtask__7(bool even_cycle, void* symtab) {
Variane_testharness.cpp:void Variane_testharness::__Vmtask__8(bool even_cycle, void* symtab) {
Variane_testharness.h:    static void __Vmtask__7(bool even_cycle, void* symtab);
Variane_testharness.h:    static void __Vmtask__8(bool even_cycle, void* symtab);
Variane_testharness.h:    static void __Vmtask__final(bool even_cycle, void* symtab);

and only see Vmtask__7 and Vmtask__8 in the generated cpp.

wsnyder, 2020-07-17:
The numbers of the tasks are arbitrary (7 and 8), but there should be typically at least two of them as there is.

Most likely the model isn't written to parallelize well, or less likely something is broken or needs tuning in the multithreading logic. You should be able to run with --debug and look at the resulting schedule *_cp.txt and graph  *_orderg_done.dot to see what ALWAYS statements Verilator thinks can be parallelized.

We unfortunately haven't put as much effort into debugging of how to get better parallelism as to the measuring and parallelism itself. So, any improvements in that would be welcome.

simdream, 2020-07-17:
but shouldn't other Mtasks in cpp file as well? From the profile, those exists, but I just dont see them in the verilated files.