verilator/verilator #2473: VL_CONSTHI_W_?X in verilated.h will bring a memory access violation.

yanx21, 2020-07-18:
> version

verilator-4.038

> testcase

```verilog
// t_consthi.v
module t(reg_i, reg_o);

input           reg_i;
output          reg_o;

wire [1:0]      reg_i;
reg  [1049:0]   reg_o;

always @ (*) begin
    if (reg_i[1] == 1'b1)
        reg_o = {986'd0, 64'hffff0000ffff0000};
    else if (reg_i[0] == 1'b1)
        reg_o = {64'hffff0000ffff0000, 986'd0};
    else
        reg_o = 1050'd0;
end

endmodule
```

After translated by verilator, we can get the following code:
```cpp
// Vt_consthi__Slow.cpp
WData/*1049:0*/ Vt_consthi::__Vtable1___Vcellout__t__reg_o[4][33];

void Vt_consthi::_ctor_var_reset() {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vt_consthi::_ctor_var_reset\n"); );
    // Body
    VL_RAND_RESET_W(1050, __Vcellout__t__reg_o);
    __Vcellinp__t__reg_i = VL_RAND_RESET_I(2);
    __Vtableidx1 = 0;
    VL_CONST_W_1X(1050,__Vtable1___Vcellout__t__reg_o[0],0x00000000);
//-
    VL_CONSTHI_W_1X(1050,1024,__Vtable1___Vcellout__t__reg_o[1],0x03fffc00); // Note: Array size is 33.
//-
    VL_CONSTLO_W_8X(768,__Vtable1___Vcellout__t__reg_o[1],0x03fffc00,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000);
    VL_CONSTLO_W_8X(512,__Vtable1___Vcellout__t__reg_o[1],0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000);
    VL_CONSTLO_W_8X(256,__Vtable1___Vcellout__t__reg_o[1],0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000);
    VL_CONSTLO_W_8X(0,__Vtable1___Vcellout__t__reg_o[1],0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000);
    VL_CONST_W_2X(1050,__Vtable1___Vcellout__t__reg_o[2],0xffff0000,0xffff0000);
    VL_CONST_W_2X(1050,__Vtable1___Vcellout__t__reg_o[3],0xffff0000,0xffff0000);
}

```

verilated.h holds the definition of `VL_CONSTHI_W_1X` 
```cpp
// verilated.h
typedef vluint32_t* WDataOutP;
#define VL_EDATASIZE 32
#define VL_WORDS_I(nbits) (((nbits) + (VL_EDATASIZE - 1)) / VL_EDATASIZE)

#define _END(obits, wordsSet) \
    for (int i = (wordsSet); i < VL_WORDS_I(obits); ++i) o[i] = 0; \
    return o

static inline WDataOutP VL_CONSTHI_W_1X(int obits, int lsb, WDataOutP obase,
                                        EData d0) VL_MT_SAFE {
    WDataOutP o = obase + VL_WORDS_I(lsb);
    o[0] = d0;
    _END(obits,1); // Note: In this case, o[0] = obase + 32, and _END() iterates another 32 times;
                           // I think it should be _END(obits - lsb, 1);
                           // Same as other VL_CONSTHI_W_?X functions.
}

```

wsnyder, 2020-07-18:
I think the fix is slightly different, but don't see how this would core dump ever as VL_WORDS(obits) should always fit within the declared size.

Please see if this works.  If not, I didn't see any core dumps even with older code so please indicate what OS and otherwise how did you get the core dump.

closed 2020-07-18 by wsnyder

yanx21, 2020-07-20:
> I think the fix is slightly different, but don't see how this would core dump ever as VL_WORDS(obits) should always fit within the declared size.
>
> Please see if this works. If not, I didn't see any [...]

Thanks, your Fix works well.
Core dump wouldn't happen in my testcase, becasuse the old code only up-overflow, and the upper momery location of `__Vtable1___Vcellout__t__reg_o` is still in data segment (is a static variable), which wouldn't cause a segment fault (core dump).
OS: Ubuntu 18.04.4 LTS
GCC version: 8.2.0