verilator/verilator #2475: Interface arrays always connect to ports in high:low order

deo248, 2020-07-20:
When an interface array port on a module is declared with low:high ordering or declared by size (which is also low:high) and connected to an interface array that is high:low ordering in a parent module, the interface array within the child module is ordered with high:low ordering, rather than having the array elements reversed. I believe they should be reversed per my understanding of the language. Additionally, Vivado treats the connections as reversed in order, which is where I noticed the simulation-synthesis mismatch.

```systemverilog
interface intf ();
    integer index;
endinterface

module subm (
    intf y[0:2] //low to high
);

endmodule
module t()
    intf test_intf[2:0](); //high downto low
    subm subm_inst (
    //here, a 2:0 signal (test_intf) is connected to a 0:2 port (y),
    //so the connections should end up reversed in ordering,
    //but I observed that within subm, they are connected 0->0, 1->1, 2->2
        .y(test_intf)
    );
endmodule
```

Using verilator git master as of today (edf6a40f1994d9a237b82e7aefeb9a83c85c2d5b).

See attached test in test_regress format. [t_interface_array_ordering.zip](https://github.com/verilator/verilator/files/4949218/t_interface_array_ordering.zip)

wsnyder, 2020-07-20:
Yes, I believe you are correct as this would match normal instance (non-interface) pin hookup "reversing".

Perhaps you could contribute a test and attempted pull request to fix this?

deo248, 2020-07-20:
I'm happy to create a pull request to integrate the attached test from above if that's how you'd like me to proceed.
However, my skillset is thoroughly in the RTL and logic design camp and I have little confidence that my mediocre C++ would be sufficient to diagnose and contribute to the elaboration code in order to fix the elaboration behavior.

wsnyder, 2020-07-20:
That's fine, we appreciate what you can help on.

closed 2021-02-21 by wsnyder