verilator/verilator #2476: port type with port > 512 bit

roccojonack, 2020-07-20:
How may we help - what is your question?

I have a 1024 bit port and the translation ends up being a sc_bv<>, but I would like to use sc_biguint<> just like for ports up to 512 bit (with switch --pins-sc-biguint). Why is this not done for larger width?

wsnyder, 2020-07-20:
I think you're asking why is there no flag to change the 512 to a different number?  I think only because no one has needed it until now - perhaps you could contribute a pull request - and update one of the tests to try it out (make a new .pl file with the new switch)?

roccojonack, 2020-07-21:
Hi, thanks. You are right. The 512 seemed like an arbitrary number and I guess you confirmed that. I believe driven by the AXI spec 1024 is definitly needed, but having a flag to control the number is indeed the better solution. I hope I'll be able to contribute a test as you suggested.

Great work that you are providing,
Rocco

roccojonack, 2020-07-22:
I looked at the tests and it seems like there is one test which checks for that limit. I could certainly extend the test to check for a higher limit. The timit itself seems to be just a "magic value" in V3AstNodes.cpp. But I don't seem to have permission to push a branch and hence also can't do a pull request...

roccojonack, 2020-07-22:
One more question on this. The modification seems fairly limited. Would you have an estimation how long it would take to a new release including this fix? Eventually I would like to integrate a fixed version into our flow, but that has to be based on a release rather than an experimental version.

wsnyder, 2020-07-22:
Yes, you could clone that test and edit the size and add the new appropriate switch to the command line.

To make a pull you should "fork" the project to your own github account, then make an edit there and push and git will say how to make a pull. There's a good description here: https://gist.github.com/Chaser324/ce0505fbed06b947d962

We usually make a release around the first week of every month - so if you got a draft in this week and leave a week for cleanups it  would align well.

roccojonack, 2020-07-22:
OK, got it. I submitted a pull request. The fix seemed fairly intuitive (and small), but since it is my first contribution you might want to check carefully. In general I was almost thinking why there is such a limit for the conversion to sc_biguint. If somebody uses the switch --pins-sc-biguint wouldn't they expect any port (>64bit) to be of type sc_biguint? At least I didn't see a good reason why not.

wsnyder, 2020-11-07:
Pull is #2487

wsnyder, 2021-03-12:
Looks like this has gone stale, if you'd like to rebase and get it in please feel free to reopen.

closed 2021-03-12 by wsnyder