verilator/verilator #2478: VCS: "Warning-[DCTL] Decimal constant too large" when compiling --protect_lib output

margej, 2020-07-24:
In my case --protect_lib output contains:
    localparam int protectlib_hash__V = 4287673301;

Which yields the following on compilation:
Warning-[DCTL] Decimal constant too large
/usr/julien/fias_core/tb/verilator/obj/verilated_riscv_cpu.sv, 337
  Decimal constant '4287673301' is too large as a 32bit signed constant, it 
  should be smaller than 2147483648.
  VCS is using converted signed value '-7293995' instead.
  To make it an unsigned constant, please use sized constant format.

The following patch resolves the issue:
```cpp
--- a/src/V3ProtectLib.cpp
+++ b/src/V3ProtectLib.cpp
@@ -197,7 +197,7 @@ private:
         // CPP hash value
         addComment(txtp, fl, "Hash value to make sure this file and the corresponding");
         addComment(txtp, fl, "library agree");
-        m_hashValuep = new AstTextBlock(fl, "localparam int protectlib_hash__V = ");
+        m_hashValuep = new AstTextBlock(fl, "localparam int protectlib_hash__V = 32'd");
         txtp->addNodep(m_hashValuep);
         txtp->addText(fl, "\n");
```

wsnyder, 2020-07-25:
Mind submitting a pull request with this - you can combine with #2479 if you want.

margej, 2020-08-04:
Trying to push a branch to raise PR from. Do I need some permissions adding?

```
[margej@vonya verilator {issue-2478|/} ]: git push --set-upstream origin issue-2478
remote: Permission to verilator/verilator.git denied to margej.
fatal: unable to access 'https://git.veripool.org/git/verilator/': The requested URL returned error: 403
```

wsnyder, 2020-08-04:
Thanks for looking into the fix, looking forward to it.

We use the fork-and-pull method so you should be pushing to your own account, see e.g. https://reflectoring.io/github-fork-and-pull/

margej, 2020-08-04:
Ok. Did wonder if I should be forking. I'm a github noob I'm affraid, having only really used git in-house previously!

margej, 2020-08-04:
PR openned, combined with #2479

wsnyder, 2020-08-04:
Merged towards 4.040 release.

closed 2020-08-04 by wsnyder