verilator/verilator #2479: Xcelium: "xmvlog: *E,NSEDGE (design.sv,5|14): Edge event 'edge' is not supported currently" on --protect_lib output

margej, 2020-07-24:
When compiling --protect_lib output with Cadence Xcelium the following error is encountered:

xmvlog: *E,NSEDGE (design.sv,6|14): Edge event 'edge' is not supported currently.
  always @(edge(clk), edge(resetn)) begin 

Suggest the following patch which should work for all simulators:

```cpp
--- a/src/V3ProtectLib.cpp
+++ b/src/V3ProtectLib.cpp
@@ -397,7 +397,7 @@ private:
         handleInput(varp);
         m_seqPortsp->addNodep(varp->cloneTree(false));
         m_seqParamsp->addText(fl, varp->name() + "\n");
-        m_clkSensp->addText(fl, "edge(" + varp->name() + ")");
+        m_clkSensp->addText(fl, "posedge " + varp->name() + " or negedge " + varp->name());
         m_cSeqParamsp->addText(fl, varp->dpiArgType(true, false) + "\n");
         m_cSeqClksp->addText(fl, cInputConnection(varp));
     }
```

wsnyder, 2020-07-25:
Mind submitting a pull request with this - you can combine with #2478 if you want.

margej, 2020-08-04:
PR openned, combined with #2478

wsnyder, 2020-08-04:
Thanks for your pull. Merged towards 4.040 release.

closed 2020-08-04 by wsnyder