verilator/verilator #2480: Possible timing race with --protect-lib output, not fully understood

margej, 2020-07-24:
I have a simulation which behaves differently with RTL than with the --protect-lib generated Verilated model and SV wrapper. 
This could be an issue in my testbench, but may also be a real issue in how the --protect-lib wrapper works. I am not yet sure, but wanted to discuss here for additional insight.

What I know is that if I replace all the blocking assignments in the generated wrapper's always block commented ...
`
// Select between combinatorial and sequential results
`
... with non-blocking assignments (so all outputs update in the same delta cycle) the issue is resolved. Clearly, this could easily be implemented as a patch as follows, but I am not sure whether or not this is the right thing to do in this case.

```cpp
--- a/src/V3ProtectLib.cpp
+++ b/src/V3ProtectLib.cpp
@@ -437,8 +437,8 @@ private:
         m_tmpDeclsp->addText(fl, " " + varp->name() + "_tmp__V;\n");
 
         m_nbAssignsp->addText(fl, varp->name() + "_seq__V <= " + varp->name() + "_tmp__V;\n");
-        m_seqAssignsp->addText(fl, varp->name() + " = " + varp->name() + "_seq__V;\n");
-        m_comboAssignsp->addText(fl, varp->name() + " = " + varp->name() + "_combo__V;\n");
+        m_seqAssignsp->addText(fl, varp->name() + " <= " + varp->name() + "_seq__V;\n");
+        m_comboAssignsp->addText(fl, varp->name() + " <= " + varp->name() + "_combo__V;\n");
         m_cComboParamsp->addText(fl, varp->dpiArgType(true, false) + "\n");
         m_cComboOutsp->addText(fl,
```

Lets consider a design where all outputs are synchronously updated using a single clock.

What happens if the TB is (rightly or wrongly) sensitive to an edge on one of these outputs, which triggers some action which has a dependency on all the other outputs.

With blocking assignments, I think the values that action will observe on these outputs will depend upon the order in which they appear in this generated sequence of blocking assignments (which I believe mirrors the declaration order on the original Verilated module)

With the RTL simulation however, all the outputs will be assigned in a single delta cycle (assuming they were assigned in a clocked always, using non-blocking assignments), and the observed state of any action triggered by one these outputs will be the previous value, for all other outputs.

I can try to put a test case together to demonstrate this.

toddstrader, 2020-07-24:
Yeah, a self-contained test for the regression suite would be handy here.  I've been using Xsim as my non-Verilator simulator of reference since it's free to download via the Vivado WebPack.  The existing --protect-lib tests know how to use Xsim so if you want to go that route hopefully you won't need to do any additional work on driver.pl.  However, it also shouldn't be too hard to extend this to support other simulators for the --protect-lib tests if you've got another simulator you'd rather test against.

margej, 2020-08-04:
Ok, thanks. I'm struggling to reproduce the issue in a standalone testcase but will keep trying ...

wsnyder, 2020-11-07:
Feel free to reopen or make new issue if have testcase.

closed 2020-11-07 by wsnyder