verilator/verilator #2485: Unsupported: The structure name is the same as the variable name

phantom-killua, 2020-07-27:
> Verilator Version
```
Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad
```
> Testcase
```systemverilog
typedef struct packed {
   logic [1:0] b1;
   logic [1:0] b2;
} t1;

typedef struct packed {
   logic [31:0] a;
   logic [31:0] b;
} t2;

typedef union packed {
   t1 t1;
   t2 t2;
} t3;

module t (
    input t3 bus,
    output logic [31:0] datao_pre
);
   always_comb datao_pre = bus.t2.a;

endmodule
```
> After translated by verilator, I met a error like this:
```
%Error: t_struct_err.v:12:7: syntax error, unexpected TYPE-IDENTIFIER
   12 |    t1 t1;
      |       ^~
%Error: t_struct_err.v:20:32: syntax error, unexpected TYPE-IDENTIFIER
   20 |    always_comb datao_pre = bus.t2.a;
      |                                ^~
%Error: Exiting due to 2 error(s)
```
> If the source code is not modified,  is there other ways to solve it？

wsnyder, 2020-07-28:
This is an unfortunate limitation of the parser that is not expected to be resolved in the near term (6 months or so), but we're working towards a more general parser eventually.

The workaround of course is to uniquely name types, e.g. have all types end in _t.

wsnyder, 2021-11-17:
Note process, semaphore and mailbox described in #3206 are another example of this problem.