verilator/verilator #2486: Optimize simple data movement

minjae63, 2020-07-27:
Some design might have long paths that only move data without any combinational logic or control flow.
Using circular queue may results in some speedup as unnecessary copy can be skipped, or even more speedup if it happens to increase cache hit rate.
This does not optimize every design, but I think it's reasonable optimization to have (at least optionally), possibly with threshold argument from user.
Following is POC design & verilator result

```
module SimpleShift(
  input         clock,
  input         reset,
  input  [31:0] io_in,
  output [31:0] io_out
);
  reg [31:0] regs_0;
  reg [31:0] regs_1;
  ...
  reg [31:0] regs_8;
  reg [31:0] regs_9;
  assign io_out = regs_0; // @[main.scala 16:64]
  always @(posedge clock) begin
    regs_0 <= regs_1;
    regs_1 <= regs_2;
    regs_2 <= regs_3;
    regs_3 <= regs_4;
    regs_4 <= regs_5;
    regs_5 <= regs_6;
    regs_6 <= regs_7;
    regs_7 <= regs_8;
    regs_8 <= regs_9;
    regs_9 <= io_in;
  end
endmodule
```

```sh
verilator --cc SimpleShift.v
```

Current result
```cpp
VL_INLINE_OPT void VSimpleShift::_sequent__TOP__1(VSimpleShift__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    VSimpleShift::_sequent__TOP__1\n"); );
    VSimpleShift* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Body
    vlTOPp->SimpleShift__DOT__regs_0 = vlTOPp->SimpleShift__DOT__regs_1;
    vlTOPp->io_out = vlTOPp->SimpleShift__DOT__regs_0;
    vlTOPp->SimpleShift__DOT__regs_1 = vlTOPp->SimpleShift__DOT__regs_2;
    vlTOPp->SimpleShift__DOT__regs_2 = vlTOPp->SimpleShift__DOT__regs_3;
    vlTOPp->SimpleShift__DOT__regs_3 = vlTOPp->SimpleShift__DOT__regs_4;
    vlTOPp->SimpleShift__DOT__regs_4 = vlTOPp->SimpleShift__DOT__regs_5;
    vlTOPp->SimpleShift__DOT__regs_5 = vlTOPp->SimpleShift__DOT__regs_6;
    vlTOPp->SimpleShift__DOT__regs_6 = vlTOPp->SimpleShift__DOT__regs_7;
    vlTOPp->SimpleShift__DOT__regs_7 = vlTOPp->SimpleShift__DOT__regs_8;
    vlTOPp->SimpleShift__DOT__regs_8 = vlTOPp->SimpleShift__DOT__regs_9;
    vlTOPp->SimpleShift__DOT__regs_9 = vlTOPp->io_in;
}
```

Expected result
```cpp
VL_INLINE_OPT void VSimpleShift::_sequent__TOP__1(VSimpleShift__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    VSimpleShift::_sequent__TOP__1\n"); );
    VSimpleShift* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Body
    vlTOPp->__OptQueue0->arr[vlTOPp->__OptQueue0->head] = vlTOPp->io_in;
    vlTOPp->__OptQueue0->head = (vlTOPp->__OptQueue0->head + 1) % vlTOPp->__OptQueue0->size;
    vlTOPp->io_out = vlTOPp->__OptQueue0->arr[vlTOPp->__OptQueue0->head];
}
```

How do you think about it? Any concerns/pitfalls in this optimization?

wsnyder, 2020-07-28:
Might you try to hand-prototype this optimization and see how much it gains? It's generally a good idea; my concern would be it might not be hot enough (enough of the total design) to result in enough gain in total for the effort.

minjae63, 2020-07-29:
Sure. thanks. will be back!