verilator/verilator #2489: Request: Transform any module that is specified in a way like this:  --top-module <module-name>

phantom-killua, 2020-07-29:
> Verilator version
```
Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad
```
> Testcase
```systemverilog
// t_module_err.v
module t (/*AUTOARG*/
   clk
   );

   input clk;
   wire           a, b, x;

   A mod_a(
           .x                           (x),
           .clk                         (clk),
           .a                           (a),
           .b                           (b));
   always @ (posedge clk) begin 
       $display("Hello World"); 
       $finish; 
   end
endmodule

module A(/*AUTOARG*/
   x,
   clk, a, b
   );
   
   input clk;
   input a, b;
   output x;
   wire   y, c;
   
   B mod_b(
           .y                           (y),
           .b                           (b),
           .c                           (c));

   C mod_c(
           .x                           (x),
           .clk                         (clk),
           .a                           (a),
           .y                           (y));
endmodule

module B(/*AUTOARG*/
   y,
   b, c
   );
   
   input b, c;
   output reg y;

   always @(*) begin
      y = b ^ c;
   end
endmodule

module C(/*AUTOARG*/
   x,
   clk, a, y
   );

   input clk;
   input a, y;
   output reg x;

   always @(posedge clk) begin
     x <= a & y;
   end
endmodule
```
> I just want to transform the module named "C" by running the following command:
```
verilator -Wall --sc --top-module C t_module_err.v
```
> But unfortunately, I meet a error like this:
```
%Warning-DECLFILENAME: t_module_err.v:1:8: Filename 't_module_err' does not match MODULE name: 't'
    1 | module t (
      |        ^
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Error: Specified --top-module 'C' isn't at the top level, it's under another cell 'A'
%Error: Exiting due to 1 error(s), 1 warning(s)
```
>  Is there any ways to solve it？

wallento, 2020-07-29:
The "proper" way is to have C in a separate file `C.sv`. I agree the error should probably be a warning, but I am not sure about the implications on the Verilator stages and about the LRM.

wsnyder, 2020-07-29:
Just to be clear, Verilator doesn't allow there to be modules that reference what you want as the top module - thus as @wallento suggests the best fix is to have C.v with "module C" and refer only to that file, alternatively remove A & B from the file and turn off the DECLFILENAME warning with -Wno-DECLFILENAME.

Closing as assume this answers your question, feel free to followup if needed.

closed 2020-07-29 by wsnyder

phantom-killua, 2020-07-30:
Sorry, maybe I don't make myself clear. My requirements are as follows: **_Supposing that N top-level models need to be generated based on N requirements for the same rtl source code._** 
To meet the preceding requirements, perform the following steps:
- Create N files that contain the path of the verilog files required for transforming. these files are named **_filelist1, filelist2, ..., filelistN_;**
- Then transform with the following commands:
**_verilator -Wall --sc --top-module module-name1 -f filelist1_**
**_verilator -Wall --sc --top-module module-name2 -f filelist2_**
**_..._**
**_verilator -Wall --sc --top-module module-nameN -f filelistN_**

But the above method is not what I want. I hope that all the models can be obtained by the command: _**verilator -Wall --sc --top-module [module-name] -f filelist, the file named "filelist" contain the path of all the rtl source code, the modules names "[module-name]" what I want as the top module.**_ In this way, I only need one _**"filelist"**_ file instead of multiple _**"filelist"**_ files.

**What do you think?**

wsnyder, 2020-07-30:
If you have N1 N2 N3 etc and all are top modules then you can pass them all in and only use --top-module to change which is modeled.

What is not supported is when N1 is a cell of N2, and you want to use N1 as a top module.  Because reading submodules might pull in other dependencies this is not something we can support at this time.  Also AFAIK no other simulator allows this.

Again, if it's possible to put one module per file, then you can simply run verilator ... N1.v and you'll get what you want.  Note Verilog-Perl has a utility that can split a file with many modules into one module per file.