verilator/verilator #2490: Tracing SystemC signals that are not inside Verilated modules

mayyxeng, 2020-07-29:
I am trying to figure out how to use a `VerilatedVcdSc` to dump vcd traces of SystemC signals that are not inside Veriated modules.
Let's say I have some Verilog module in `decoder.v`, and a handwritten SystemC module `trigger.h`. The Verilog module `decoder.v` is Ventilated into a SystemC module and is connected to the `trigger.h` SystemC module in a third SystemC module in `network.h`. Now I want to trace all the signals, i.e. the signals in the Verilated `decoder.v` and the signals from the handwritten `trigger.h` module.

To dump traces in SystemC, I use the `sc_trace( sc_trace_file* tf, const sc_signal_in_if<T>& object, const char* name )`. But I suppose I can not use this on a `VerilatedVcdSc*` that is used to capture all the signals in `decoder.v`.

How can I have a mix of traditional SystemC traces and the Verilated traces at the same time?

Thanks!

wsnyder, 2020-07-29:
VerilatedVcdSc is a sc_trace_file, so you should be able to pass that into sc_trace and it'll all trace together.

BTW it would probably be nice if examples/make_tracing_sc and cmake_tracing_sc showed how this worked, if you'd like to edit those to show it and make a pull request it would be appreciated.

Closing as presume answers your problem, feel free to still comment.

mayyxeng, 2020-07-30:
Thanks for the tip, so I have tried passing a pointer of type `VerilatedVcdSc` to the SystemC's `trace(..)` function. However, I get a compilation error saying:
```
 error: ‘sc_core::sc_trace_file’ is an inaccessible base of ‘VerilatedVcdSc’
```
I am using GCC 7.5.0 on Ubuntu 18.04 and Verilator 4.038. Looking at `verilated_vcd_sc.h` the error makes sense since Sine `VerilatedVcdSc` is inheriting `sc_trace_file` privately:
```c++
class VerilatedVcdSc : sc_trace_file, public VerilatedVcdC {
    // CONSTRUCTORS
    VL_UNCOPYABLE(VerilatedVcdSc);
```
I simply made the inheritance public to try my luck. I modified the `sc_main` in `examples/make_tracing_sc` to include a non-Verilated signal `sc_signal<uint32_t> ticks_counter` and called `sc_trace(tfp, ticks_counter, "ticks_counter")` right after opening the trace file:
```c++
if (flag && 0 == strcmp(flag, "+trace")) {
    cout << "Enabling waves into logs/vlt_dump.vcd...\n";
    tfp = new VerilatedVcdSc;
    top->trace(tfp, 99);  // Trace 99 levels of hierarchy
    Verilated::mkdir("logs");
    tfp->open("logs/vlt_dump.vcd");
    sc_trace(tfp, ticks_counter, "ticks_counter"); // the new line added
}
```
The program compiles and dumps traces for the Verilated module but _does not_ seem to be registering the `ticks_counter`. I dug into the code a bit and I realized that in `VerialtedVcd` there is a `NameMap* m_namemap` field used to hold the signals being traces. Contrary to this, SystemC's `vcd_trace_file` has a field `std::vector<vcd_trace*> traces` and this vector holds the signals to be traced. The vector is populated by the overridden calls to `sc_trace_file::trace(...)`. However, all these methods in the `VerilatedVcdSc` are simply empty therefore the signals will not be registered. I think this is the reason why the inheritance is from `sc_trace_file` is kept private.
To my understanding, unlike the SystemC way of calling `sc_trace(...)`, signals are registered for tracing using the following methods defined in `VerilatedVcdC` class:
```c++
    //=========================================================================
    // Internal interface to Verilator generated code

    void declBit(vluint32_t code, const char* name, bool array, int arraynum);
    void declBus(vluint32_t code, const char* name, bool array, int arraynum, int msb, int lsb);
    void declQuad(vluint32_t code, const char* name, bool array, int arraynum, int msb, int lsb);
    void declArray(vluint32_t code, const char* name, bool array, int arraynum, int msb, int lsb);
    void declDouble(vluint32_t code, const char* name, bool array, int arraynum);

```
However, this seems to be meant for generated code, not manually written code. And even if this is mean to be used by manual code, I do not know how to use them.

I am new to both Verilator and SystemC, I would appreciate some suggestions on how I can get around this. 

Thanks in advance.

wsnyder, 2020-07-30:
Ah, yes, you're right.  Unfortunately I don't see a solution other than making two trace files.  Also unfortunately you can't just concatenate two VCDs to combine them.  However, the viewers I'm aware of allow reading more than one trace file in the same session, so hopefully that works for you.

mayyxeng, 2020-07-30:
I see, thank you. I close this issue then.

closed 2020-07-30 by mayyxeng