verilator/verilator #2491: Unsupported: The multidimensional array defined parameter is assigned with '{default:0}

phantom-killua, 2020-07-30:
> Verilator version:
```
Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad
```
> Testcase
```systemverilog
//t_param_err.v
/* verilator lint_off DECLFILENAME */
module t;
/* verilator lint_on DECLFILENAME */
parameter p [1:0][7:0][7:0] = '{default:0};
initial begin
if (p[0][0][0] != 1'b0) $stop;
$write("*-* All Finished *-*\n");
$finish;
end
endmodule
```
> Then translated by verilator with the following  command: _**verilator -Wall --sc t_param_err.v**_ , I meet a error like this:
```
%Error: t_param_err.v:7:15: Illegal assignment of constant to unpacked array
                          : ... In instance t
    7 |       if (p[0][0][0] != 1'b0) $stop;
      |               ^
%Error: Exiting due to 1 error(s)
```
> If the following statement is modified in this way, translated successfully by verilator.
```systemverilog
parameter p [1:0][7:0][7:0] = '{default:0}; 
```
> changed to
```systemverilog
parameter p [1:0][7:0][7:0] = '{default:'{default:'{default:0}}}; 
```
> I think it might be a bug, what's your opinion?

wsnyder, 2020-08-02:
This does seem to be legal syntax.  There's both packed and unpacked versions of this that should be legal.

Perhaps you could attempt to teach V3Width how to support this?

Here's the modifications to the tests for this:

```
--- a/test_regress/t/t_array_pattern_packed.v
+++ b/test_regress/t/t_array_pattern_packed.v
@@ -48,6 +48,10 @@ module t (/*AUTOARG*/
       // However IEEE suggests this is legal.
       //array_simp = '{2{ '{2{ 4'd3, 4'd2 }} }};  // Note it's not '{3,2}

+      // Default
+      array_simp = '{default:1};
+      if (array_simp !== 32'h1111_1111) $stop;
+
       $write("*-* All Finished *-*\n");
       $finish;
    end
diff --git a/test_regress/t/t_array_pattern_unpacked.v b/test_regress/t/t_array_pattern_unpacked.v
index aaf1db6c4..2a6bd8c34 100644
--- a/test_regress/t/t_array_pattern_unpacked.v
+++ b/test_regress/t/t_array_pattern_unpacked.v
@@ -52,6 +52,11 @@ module t (/*AUTOARG*/);
       if ({irep[2][1], irep[2][2], irep[2][3], irep[2][4], irep[2][5], irep[2][6]}
           != {32'h4, 32'h5, 32'h4, 32'h5, 32'h4, 32'h5}) $stop;

+      // Default
+      irep = '{default:1};
+      if ({irep[1][1], irep[1][2], irep[1][3], irep[1][4], irep[1][5], irep[1][6]}
+          != {32'h1, 32'h1, 32'h1, 32'h1, 32'h1, 32'h1}) $stop;
+
       $write("*-* All Finished *-*\n");
       $finish;
    end

```

wsnyder, 2022-01-01:
Ignore that last commit message - meant to refer to #2431 related to force/release.

ArturBieniek4, 2026-09-02:
Was fixed by #7202, testcase works on current master.

closed 2026-09-02 by ArturBieniek4