verilator/verilator #2493: Unsupported: Type conversion explicitly

phantom-killua, 2020-07-30:
> Verilator version
```
Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad
```
Testcase
```systemverilog
// t_type_conversion_err.v
/* verilator lint_off DECLFILENAME */
module t(
  in1,
  in2,
  out
  );
/* verilator lint_on DECLFILENAME */
  input logic in1;
  input logic [1:0] in2;
  output logic [31:0] out;
  
  typedef logic[7:0] data;
  
  /* verilator lint_off WIDTH */
  assign out = data'(in1 << in2);
  /* verilator lint_on WIDTH */
  
endmodule
```
> After translated by verilator with the command:  _**verilator -Wall --sc --top-module t t_type_conversion_err.v**_, we can get the following code:
```c++
// Vt.cpp
VL_INLINE_OPT void Vt::_combo__TOP__1(Vt__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vt::_combo__TOP__1\n"); );
    Vt* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Body
    VL_ASSIGN_ISI(2,vlTOPp->__Vcellinp__t__in2, vlTOPp->in2);
    VL_ASSIGN_ISI(1,vlTOPp->__Vcellinp__t__in1, vlTOPp->in1);
    VL_ASSIGN_SII(32,vlTOPp->out, (1U & ((IData)(vlTOPp->__Vcellinp__t__in1) 
                                         << (IData)(vlTOPp->__Vcellinp__t__in2))));
}
```
> I think that the following statement is wrong, what's your opinion?
```c++
VL_ASSIGN_SII(32,vlTOPp->out, (1U & ((IData)(vlTOPp->__Vcellinp__t__in1) 
                                         << (IData)(vlTOPp->__Vcellinp__t__in2))));
```

wsnyder, 2020-07-30:
I'm not seeing what's wrong.

Note Verilator requires all inputs have any unspecified upper bits be zero, so in2 can't ever be more then 3.  If you compile with "-DVL_DEBUG=1" it should warn of this.

flex-liu, 2020-08-04:
Verilator output result bit width is wrong. The expected output bit width of the forced type conversion is 8, and the result output bit width is 1.

wsnyder, 2020-08-04:
I still think the logic is correct, do you get the wrong value - can you send a complete test case that works in another simulator and fails in Verilator?  A one bit signal extended arbitrarily is still 1 bit of data (the "1U &") and signals 2-32 bits in SC are always 32 bits.

flex-liu, 2020-08-05:
Yes, ARM's CMS tool handles it differently from verilator. In order to solve this problem, we write verilog like this"assgin out = in1<< in2;", and verilator can handle it correctly. The generated code is like this("0xffU &").

wsnyder, 2020-08-10:
Understood now, basically casts shouldn't be self-determined expressions, even though IEEE doesn't seem to say one way or the other.

Fixed in git towards 4.040 release.

closed 2020-08-10 by wsnyder