verilator/verilator #2502: Build failure on OSX due to missing C++11 flag

kkiningh, 2020-08-17:
Attempting to build the most recent commit of Verilator fails on OSX due to C++11 issues (trace below). Note that configure correctly picked up that my version of clang has C++11 support. Adding the flag manually using `CFLAGS='std=c++11' make` results in a working build.

```
kkiningh:~/Workspace/Research/verilator$ git rev-parse HEAD
ea9b65fe6daa8b7f1546330b473aad308dc95764
kkiningh:~/Workspace/Research/verilator$ make
autoconf
./config.status --recheck
running CONFIG_SHELL=/bin/sh /bin/sh ./configure --no-create --no-recursion
configuring for Verilator 4.099 devel
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for a BSD-compatible install... /usr/bin/install -c
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
compiler is g++ --version = Apple clang version 11.0.0 (clang-1100.0.33.16)
checking that C++ compiler can compile simple program... yes
checking for perl... /usr/bin/perl
checking for flex... /usr/bin/flex
/usr/bin/flex --version = flex 2.5.35 Apple(flex-32)
checking for bison... /usr/bin/bison
/usr/bin/bison --version = bison (GNU Bison) 2.3
checking for ccache... no
checking how to run the C++ preprocessor... g++ -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for size_t... yes
checking for size_t... (cached) yes
checking for inline... inline
checking whether g++ accepts -std=gnu++14... yes
checking whether g++ accepts -std=c++03... yes
checking whether g++ accepts -Wextra... yes
checking whether g++ accepts -Wfloat-conversion... yes
checking whether g++ accepts -Wlogical-op... no
checking whether g++ accepts -Wthread-safety... yes
checking whether g++ accepts -Qunused-arguments... yes
checking whether g++ accepts -faligned-new... yes
checking whether g++ accepts -Wno-unused-parameter... yes
checking whether g++ accepts -Wno-shadow... yes
checking whether g++ accepts -Wno-char-subscripts... yes
checking whether g++ accepts -Wno-null-conversion... yes
checking whether g++ accepts -Wno-parentheses-equality... yes
checking whether g++ accepts -Wno-unused... yes
checking whether g++ accepts -faligned-new... yes
checking whether g++ accepts -fbracket-depth=4096... yes
checking whether g++ accepts -fcf-protection=none... yes
[... 120 lines cut ...]
../V3Error.h:132:15: warning: expression result unused [-Wunused-value]
        : m_e{EC_MIN} {}
              ^~~~~~
../V3Error.h:138:14: error: expected '('
        : m_e{static_cast<en>(_e)} {}
             ^
../V3Error.h:138:34: error: expected ';' after expression
        : m_e{static_cast<en>(_e)} {}
                                 ^
                                 ;
../V3Error.h:138:15: warning: expression result unused [-Wunused-value]
        : m_e{static_cast<en>(_e)} {}
              ^~~~~~~~~~~~~~~~~~~
../V3Error.h:213:22: error: no member named 'ascii' in 'V3ErrorCode'
    return os << rhs.ascii();
                 ~~~ ^
../V3Error.h:242:12: error: unknown type name 'constexpr'
    static constexpr unsigned MAX_ERRORS = 50;  // Fatal after this may errors
           ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
5 warnings and 20 errors generated.
make[2]: *** [Verilator.o] Error 1
make[1]: *** [../bin/verilator_bin_dbg] Error 2
make: *** [verilator_exe] Error 2
```

wsnyder, 2020-08-17:
Committed attempted fix... let me know if has a problem, otherwise the nightly OSX builds should validate it.

kkiningh, 2020-08-17:
Awesome, that seems to have fixed it!

Minor side note - there's a lot of warnings generated when compiling with clang in C++11 mode due to the `register` keyword in the generated lexer. Might be useful to pass `-Wno-deprecated-register` or stick a `define register ` at the top of verilog.l.

Let me know if you want me to open another issue for this.

closed 2020-08-17 by kkiningh