verilator/verilator #2510: The generated code in perticular cases shall be incorrect when the reg is assigned by both block and non-block statements.

yanx21, 2020-08-24:
My testcase
```verilog
// t_blknblk.v
module t (
    clk
);
    input     clk;

    reg       i_pre;
    reg       i_post;
    reg       o_pre;
    reg       o_post /*verilator public_flat_rw*/;
    reg [1:0] ff;

    always @ (posedge clk) begin
        i_post  <= i_pre;
    end
    always @ (*) begin
        ff[0]   =  i_post;
    end
    always @ (posedge clk) begin
        ff[1]   <= ff[0];
        o_pre   <= ff[1];
        o_post  <= o_pre;
    end
endmodule
```
  Verilator reports BLKANDNBLK warning as compiling the above rtl code. but I insist the rlt code is pretty normal, so I turn it off.
  The following cpp code is generated by verilator-4.040 with `--cc --Wno-BLKANDNBLK`
```cpp
// obj_dir/Vt_blknblk.cpp
// ...
VL_INLINE_OPT void Vt_blknblk::_sequent__TOP__1(Vt_blknblk__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vt_blknblk::_sequent__TOP__1\n"); );
    Vt_blknblk* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Variables
    CData/*1:0*/ __Vdly__t__DOT__ff;
    // Body
    __Vdly__t__DOT__ff = vlTOPp->t__DOT__ff;
    vlTOPp->t__DOT__i_post = vlTOPp->t__DOT__i_pre;
    vlTOPp->t__DOT__o_post = vlTOPp->t__DOT__o_pre;
    __Vdly__t__DOT__ff = ((1U & (IData)(__Vdly__t__DOT__ff))
                          | (2U & ((IData)(vlTOPp->t__DOT__ff)
                                   << 1U)));
    vlTOPp->t__DOT__o_pre = (1U & ((IData)(vlTOPp->t__DOT__ff)
                                   >> 1U));
    vlTOPp->t__DOT__ff = ((2U & (IData)(vlTOPp->t__DOT__ff))
                          | (IData)(vlTOPp->t__DOT__i_post)); // <1>
    vlTOPp->t__DOT__ff = __Vdly__t__DOT__ff;                  // <2>
}
// ...
```
  As you can see above, the order of codes noted by  `<1>` and `<2>` are incorrect.  
  The previous assignment`<1>` is overwritten by the following assignment`<2>`.  

---

  In my opinion, `V3Order` leads to this problem when 2 conditions are met.
  1. `<1>` has smaller ranking result than `<2>`
  2. LogicVertex of `<1>` is moved to a sequential domain.

  In `V3Order`, it seems that you didn't consider the dependency between `COMBO LogicVertex` and `SEQUENT LogicVertex` when they drive the same `VarLogic`, while building the graph.

  I think this is definitely a issue. Can you fix this in a later version？Or give me some advices?

  Thanks a lot.

wsnyder, 2020-08-24:
As the documentation says : "Verilator reports this as an error, because ignoring this warning may make Verilator simulations differ from other simulators."  There's an example of how to recode this.  Another possibility may be the new /*verilator split_var*/ attribute on the signal, if that works let us know or ideally give us a patch to update the documentation.

This is unlikely to be fixed in the near/medium term, and is covered by other bugs already, so just marking this one duplicate.

closed 2020-08-24 by wsnyder

yanx21, 2020-08-26:
As you said, other simulators report the warning, but they can handle this case correctly anyway.

BTW, what do you mean about "covered by other bugs already" ?

Thanks for your reply.

wsnyder, 2020-08-26:
A mix of #63, #449, #648 perhaps others.  Fundamentally the problem is Verilator currently orders and schedules based on complete signals, not bits.