verilator/verilator #2514: Fix cmake for C++11

wsnyder, 2020-08-24:
The code base has been updated to C++11, but some cmake tests are failing in Travis cron, due to what seems the std not getting properly set. The rule in the gnumake autoconf is we try a build without any compiler version, and if that fails we try a std= flag, which seems to work well, so presumably a similar rule is needed for the cmake rules.

Can someone that knows the cmake magic look into this please? @patstew, @toddstrader, or @wallento maybe?

Note I'm still working through other failures so cron isn't otherwise clean yet.

Thanks

wsnyder, 2020-08-27:
Made an attempted fix, we'll see tonight how it does.

patstew, 2020-08-27:
Not sure about the autotools, but you probably want a `set(CMAKE_CXX_STANDARD 11)` in the relevant `CMakeLists.txt`

wsnyder, 2020-08-27:
Could make a you pull for this?

wsnyder, 2020-08-28:
Passed last night, thanks.

closed 2020-08-28 by wsnyder