verilator/verilator #2515: Is it possible to generate an Android library (.so) from the Verilator output?

Icaro-Lima, 2020-08-25:
I spent all night trying to link Verilator output and generate a .so file using Android NDK. The problem is because I won't have much experience, and the Android NDK documentation is not clear.

wsnyder, 2020-08-25:
I don't personally have Android experience and haven't heard of someone doing this, but maybe the community can assist.

If you figure it out please post back a recipe to help the next person attempting this, thanks.

wallento, 2020-08-25:
I have a bit of NDK experience, what issue are you facing?

Icaro-Lima, 2020-08-26:
I managed to compile the code generated by verilator for Android.

Here is the CMakeLists.txt I used:
``` cmake
cmake_minimum_required(VERSION 3.8)
project(cmake_simulator)

set(CMAKE_SYSTEM_NAME Android)
set(CMAKE_SYSTEM_VERSION 21)
set(CMAKE_ANDROID_ARCH_ABI x86)
set(CMAKE_ANDROID_NDK /home/icarolima/Android/Sdk/ndk/21.3.6528147)
set(CMAKE_ANDROID_STL_TYPE gnustl_static)

set(CMAKE_TOOLCHAIN_FILE /home/icarolima/Android/Sdk/ndk/21.3.6528147/build/cmake/android.toolchain.cmake)

find_package(verilator HINTS $ENV{VERILATOR_ROOT} ${VERILATOR_ROOT})
if (NOT verilator_FOUND)
  message(FATAL_ERROR "Verilator was not found. Either install it, or set the VERILATOR_ROOT environment variable")
endif()

# Create a new executable target that will contain all your sources
add_library(simulator SHARED simulator.cpp)

# Add the Verilated circuit to the target
verilate(simulator
  INCLUDE_DIRS "."
  SOURCES top.sv
  VERILATOR_ARGS -Wno-CASEINCOMPLETE -Wno-WIDTH -Wno-COMBDLY -cc +1800-2012ext+sv)
```

I sincerely have not tested the libsimulator.so yet but I will do it soon and update this thread.

Icaro-Lima, 2020-08-27:
Here is the final version that worked very well on my Android Moto G6 Play!

```
# CMakeLists.txt

cmake_minimum_required(VERSION 3.8)
project(cmake_simulator)

list(APPEND CMAKE_FIND_ROOT_PATH /usr/share)

find_package(verilator HINTS $ENV{VERILATOR_ROOT} ${VERILATOR_ROOT})
if (NOT verilator_FOUND)
  message(FATAL_ERROR "Verilator was not found. Either install it, or set the VERILATOR_ROOT environment variable")
endif()

# Create a new executable target that will contain all your sources
add_library(simulator SHARED simulator.cpp)

# Add the Verilated circuit to the target
verilate(simulator
  INCLUDE_DIRS "."
  SOURCES top.sv
  VERILATOR_ARGS -Wno-CASEINCOMPLETE -Wno-WIDTH -Wno-COMBDLY -cc +1800-2012ext+sv)
```

And the commands I use to compile:
```
mkdir build

cd build && \
	cmake .. \
	-DCMAKE_TOOLCHAIN_FILE=${NDK}/build/cmake/android.toolchain.cmake \
        -DANDROID_ABI=armeabi-v7a \
        -DANDROID_NATIVE_API_LEVEL=21

cd build && cmake --build .
```

closed 2020-08-27 by Icaro-Lima