verilator/verilator #2517: FreeBSD with C++11 failing t_hier_block_prot_lib

wsnyder, 2020-08-25:
The tests t_hier_block_vlt and t_hier_block_prot_lib seem to be failing since we started C++11, see e.g. https://travis-ci.com/github/verilator/verilator/jobs/377453221

I suspect there's a mis-match of compiler flags so the objects aren't compatible with linking, @yTakatsukasa can you take a look please, thanks

yTakatsukasa, 2020-08-25:
Oh, I did not know that. Thank you letting me know.

I will take a look at it starting from installing FreeBSD to VM.

closed 2020-08-26 by yTakatsukasa

yTakatsukasa, 2020-08-27:
[Travis](https://github.com/verilator/verilator/runs/1036101604) shows that tests on FreeBSD passes after PR #2521  is merged.

Tests on mac seems to fail on the similar way, but I can not access mac. Need to guess what's going on.

gezalore, 2020-08-27:
If you fancy, Travis Debug mode works on the Mac (well, it did last time I tried):
https://docs.travis-ci.com/user/running-build-in-debug-mode/#restarting-a-job-in-debug-mode-via-api
Once you re-start job in debug mode with the API, it should print an SSH line in he on-line log you can use to log in. I usually run:
> travis_run_before_install
> travis_run_install
> travis_run_before_script

to get me to a working environment, though depending on where the problem is you might need different steps. Keep in mind the 50 minute Travis timelimit still applies, so it will kick you out afterwards. Let me know if you would like more detail.

PR #2521 (2020-08-26): Fix test failure on FreeBSD
Fixes #2517
I installed FreeBSD-12.1 to VM and confirmed the symptom.
The test failed because the archiver used on FreeBSD is not gnu one, so thin-archive is not supported on the platform.
Thin-archive was used to merge multiple .a files, but turns out to be binutils specific.

This PR uses extract and archive options of ar command that are common operation among platforms.

I confirmed t_hier_block_*.pl passed on FreeBSD 12.1 on VM.

-time option is removed not to see warnings from C++ compiler during test.