verilator/verilator #2518: Provide Getting Started notes, examples and design patterns for Verilator

mulderp, 2020-08-25:
How may we help - what is your question?

Hello, I would like to use Verilator (or the combination of C++ and Verilog) to better understand digital system design. My original question was formulated in this issue actually https://github.com/verilator/verilator/issues/2056#issuecomment-679965700 

Now in the meanwhile I have started a gitbook repo: https://github.com/mulderp/verilator_book

Questions I would like to explore are:
1) What parts of a testbench and system are better written in C++ ? What parts in Verilog?
2) How can Verilator be used to improve debugging a design (trace signals, do validation, ... )
3) How can Verilator be used to improve simulation performance?
4) What other use cases are there to have C++ code mix with Verilog?

Any other topics that I don't see right now ... ? Maybe sharing/packaging a design, ... ?

Thanks for your interest and pull requests!

wsnyder, 2020-08-26:
Thanks for your work on this, looking forward to watching the progress. Marking help wanted so people can comment/review.

mulderp, 2020-08-29:
I did some updates on using an example with CMakeLists.txt and SystemC with a clock https://mulder-patrick.gitbook.io/verilator/basics

However, the clock syntax I use from the man page gives:

```
$ ./Vhello

        SystemC 2.3.4_pub_rev_20191203-Accellera --- Aug 29 2020 13:00:14
        Copyright (c) 1996-2019 by all Contributors,
        ALL RIGHTS RESERVED

Info: (I804) /IEEE_Std_1666/deprecated:
    sc_clock(const char*, double, double, double, bool)
    is deprecated use a form that includes sc_time or
    sc_time_unit
```

How would we define the sc clock in the testbench? Thanks!

wsnyder, 2020-08-29:
Updated documentation - please see the examples/ sc files for bigger clean examples.

mulderp, 2020-08-30:
Thanks yes, the examples look good. Probably I explore those without SystemC first. Thanks!

closed 2021-02-21 by wsnyder