verilator/verilator #2525: Queue within classes

nanduraj1, 2020-08-29:
Thanks for taking the time to report this.

What would you like added/supported?

enable `object.queue.push_back()`. Currently it gives
```
%Error-UNSUPPORTED: t.sv:25:19: Unsupported: Non-variable on LHS of built-in method 'push_back'
                                   : ... In instance tb_top
   25 |         c2[0].que.push_back(10);  
      |                   ^~~~~~~~~
%Error: Exiting due to 1 error(s)
```

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
```
module t();
    class Cls;
        int que[$];

        task push_data(int val);
            que.push_back(val);
        endtask
    endclass

    initial begin
        Cls c2 [1:0];

        c2[0] = new();

        c2[0].push_data(20);   // Works

        $display("size: %d", c2[0].que.size()); // Works

        c2[0].que.push_back(10); // Unsupported
    end

endmodule
```
May we assist you in trying to fix this yourself?
yes

wsnyder, 2020-08-29:
Thanks for the report, fixed.

closed 2020-08-29 by wsnyder

nanduraj1, 2020-08-29:
Thanks @wsnyder .