verilator/verilator #2529: If block with $fatal call executes when it looks like it shouldn't

jackkoenig, 2020-09-02:
Here is a reproducible example in a repository (Apache 2.0 licensed code taken from Rocket Chip): https://github.com/jackkoenig/verilator-fatal-issue#setup

I am very sorry that the example is so much Verilog and has additional dependencies on `riscv-isa-sim`. I tried very hard to reduce the test case further, including creating a testbench with just the problematic module (`TLMonitor_31`), but I could not reproduce the bug with just the one module. 

However, I did remove every single `$fatal` call except the 2 that seem related to the problem, so I hope it's reduced enough that the problem isn't that hard to see. I also deleted all of the extra `TLMonitors` so there's just the 1.

I saw the same behavior on several Verilator versions including git master: https://github.com/jackkoenig/verilator-fatal-issue#additional-information

If this is a bug, I primarily need to understand what causes it to manifest so that FIRRTL can emit Verilog that avoids it.

wsnyder, 2020-09-02:
Can you update the .git with any needed binaries please?

configure: error: device-tree-compiler not found

jackkoenig, 2020-09-02:
Yes of course, sorry about that

jackkoenig, 2020-09-02:
I added standard package manager install information for Ubuntu and MacOS for `device-tree-compiler`: https://github.com/jackkoenig/verilator-fatal-issue/blob/master/README.md#less-common-dependencies

Is that sufficient?

wsnyder, 2020-09-04:
Appears to be a bug in the gate optimizer as the -Og flag works around this.  Given the size will take some time to diagnose.

wsnyder, 2020-09-04:
SimDTM.v has code like this:

```
  assign #0.1 debug_req_valid = __debug_req_valid;
  assign #0.1 debug_req_bits_addr = __debug_req_bits_addr[6:0];
  assign #0.1 debug_req_bits_op = __debug_req_bits_op[1:0];
  assign #0.1 debug_req_bits_data = __debug_req_bits_data[31:0];
  assign #0.1 debug_resp_ready = __debug_resp_ready;
  assign #0.1 exit = __exit;

  always @(posedge clk)
```

Verilator ignores these delays, and warns about this, but you suppressed the warning. Therefore this code races.  Using negedge avoids the race and makes the test pass (You can then also remove the #delays which is much better coding.)

closed 2020-09-04 by wsnyder