verilator/verilator #2533: cmake-based verilated project rebuilds with each call of make

yurivict, 2020-09-06:
I followed the documentation https://www.veripool.org/wiki/verilator/Manual-verilator#CMake, but every time I run the 'make' command it rebuilds all C++ again and again.

This isn't supposed to happen normally during the make command run. It should rebuild the target when needed and do nothing when called again, until some sources change.

yurivict, 2020-09-10:
After a round of debugging, I think that cmake is a culprit: https://gitlab.kitware.com/cmake/cmake/-/issues/21181

yurivict, 2020-09-10:
This cmake custom_command
```
  add_custom_command(OUTPUT ${GENERATED_SOURCES} "${VCMAKE}"
                     COMMAND ${VERILATOR_COMMAND}
                     WORKING_DIRECTORY "${CMAKE_CURRENT_SOURCE_DIR}"
                     DEPENDS "${VERILATOR_BIN}" ${${VERILATE_PREFIX}_DEPS} VERBATIM)
```
is supplied outputs in ```${GENERATED_SOURCES}``` that are never generated.

For example, my top module is called ```mymod.v```, and ```Vmymod__22__Slow.h``` is in ```GENERATED_SOURCES```. ```Vmymod__22__Slow.cpp```exists, but its .h is missing.

yurivict, 2020-09-10:
This section
```
  foreach(GENERATED_C_SOURCE ${GENERATED_C_SOURCES})
    get_filename_component(C_OUTPUT_NAME_WE "${GENERATED_C_SOURCE}" NAME_WE)
    if(C_OUTPUT_NAME_WE MATCHES ".*Trace.*")
      continue()
    endif()
    list(APPEND GENERATED_H_SOURCES "${VDIR}/${C_OUTPUT_NAME_WE}.h")
  endforeach()
```
adds .h files for all .cpp files, except for ones that contain the string ```Trace```, but many other .cpp files don;t have corresponding .h files too.

Workaround: comment out this line:
> list(APPEND GENERATED_H_SOURCES "${VDIR}/${C_OUTPUT_NAME_WE}.h")

@wsnyder Do you know how to build a correct header file list from .cpp file list? Or, otherwise, what is the correct fix of this problem?

wsnyder, 2020-09-10:
Verilator already makes a dependecy list with all files (.cpp and .h) for make. Can you use that, filtering for .h, or are you suggesting what Verilator already makes that way is used for cmake but wrong (extra files?)

yurivict, 2020-09-11:
It adds bogus .h files that don't get created by the subsequent verilator command.

How to build a correct .h list?

wsnyder, 2020-09-11:
It looks like .cmake makes the .h from the .c names in verilator-config.cmake.in. It should be taking it from verilator output generated by walking the list of generated files similar to how V3File::writeDepend writes them for gmake.

yurivict, 2020-09-11:
Maybe it's not necessary to add header files there? Because .cpp and .h are generated at the same time with the same command, they are either all outdated or all up to date.

wsnyder, 2020-09-11:
That's true.

yurivict, 2020-09-11:
Then removing the above section generating ```GENERATED_H_SOURCES``` should fix the problem.

wsnyder, 2020-09-11:
Can you give that fix a try please and if it works submit a patch? Thanks.

yurivict, 2020-09-11:
https://github.com/verilator/verilator/pull/2543

wsnyder, 2020-09-11:
Fixed with #2543, thanks!

closed 2020-09-11 by wsnyder