verilator/verilator #2536: Dual clock, dual run testbench freezes

bluecmd, 2020-09-08:
Hi,

I'm using Verilator 4.100 with cocotb 1.4.0.
The cocotb bug is https://github.com/cocotb/cocotb/issues/2051 and has some debugging by the maintainers but no workaround. It was suggested this is likely a Verilator issue, so I'm filling this bug report.

I have made this minimal repro: https://gist.github.com/bluecmd/fe70d5279bfcaca470c6556ccd502933.

- I set up two identical tests creating two clocks and one AvalonSTMonitor
- If I run a single test using `make TESTCASE=test_a` either test works.
- If I make the test only use a single clock, running both tests works.
- If I initialize the AvalonSTMonitor before calling await on anything both tests work.
- Using `make SIM=icarus` works fine.

As can be seen in the debug log in the other bug cocotb claims to return the control to Verilator and then nothing happens.

@ktbarrett for visibility.

wsnyder, 2020-09-08:
@elgorwi would you mind taking a look please?

elgorwi, 2020-09-11:
Looking at the linked issues, this seems to be related to the cocotb scheduler, which is something I’m not familiar with.

Not sure when I’ll have time to look at this in detail, but perhaps @wallento may be able to help out?

ktbarrett, 2020-09-11:
I'm not so sure it's cocotb's scheduler; the same code also works in other simulators, like Icarus. The hanging usually occurs with verilator because cocotb returns to the simulator from the Python context with no events scheduled (other simulators detect this and fail). However, in this case, there are clearly events being registered before returning from Python. I'm sure it's some assumption being made about the registering of events in the VPI. Whether it's cocotb or verilator (or both) I'm not yet certain.

I'll work on trying to recreate the problem with a simpler example.

markusdd, 2020-10-05:
Hi, we just had another occurence of this issue, and it looks very much related:
https://github.com/cocotb/cocotb/issues/2096#

We have 2 scenarios: After forking 2 coroutines that write and read from a FIFO interface, when waiting for a number of clocks, all 1000 edges are triggered immidiately without advancing sim time.
When using a timer instead for 10000 us, the simulation hangs up after handing control back to the simulator.
It hangs so bad, I have to kill it from htop, will not respond to Ctrl+C .

The log files for both scenarios can be found in the linked ticket.

If anyone from verilator needs access to the project I can assign them to a privat mirror here on Github, just let me know.

markusdd, 2020-10-05:
Hi, in order to simplify things, I made a small example repo we can use to get to the bottom of this:

https://github.com/markusdd/verilator_coco_debug

markusdd, 2020-10-05:
I did some debugging:

```
(gdb) bt
#0  0x000000000042ba51 in VerilatedVpiImp::cbReasonRemove(VerilatedVpioCb*) ()
#1  0x0000000000424166 in vpi_remove_cb ()
#2  0x00007fefad9ba6b1 in virtual thunk to VpiValueCbHdl::cleanup_callback() () at /usr/include/c++/10/ext/new_allocator.h:103
#3  0x00007fefad9a2dea in GpiValueCbHdl::run_callback (this=0x1145748) at cocotb/share/lib/gpi/GpiCbHdl.cpp:173
#4  0x00007fefad9b6def in handle_vpi_callback (cb_data=<optimized out>) at cocotb/share/lib/vpi/VpiImpl.cpp:583
#5  0x000000000042bf83 in VerilatedVpiImp::callValueCbs() ()
#6  0x00000000004209f3 in VerilatedVpi::callValueCbs() ()
#7  0x0000000000404a33 in main ()
(gdb) bt
#0  0x000000000042ba51 in VerilatedVpiImp::cbReasonRemove(VerilatedVpioCb*) ()
#1  0x0000000000424166 in vpi_remove_cb ()
#2  0x00007fefad9ba6b1 in virtual thunk to VpiValueCbHdl::cleanup_callback() () at /usr/include/c++/10/ext/new_allocator.h:103
#3  0x00007fefad9a2dea in GpiValueCbHdl::run_callback (this=0x1145748) at cocotb/share/lib/gpi/GpiCbHdl.cpp:173
#4  0x00007fefad9b6def in handle_vpi_callback (cb_data=<optimized out>) at cocotb/share/lib/vpi/VpiImpl.cpp:583
#5  0x000000000042bf83 in VerilatedVpiImp::callValueCbs() ()
#6  0x00000000004209f3 in VerilatedVpi::callValueCbs() ()
#7  0x0000000000404a33 in main ()
(gdb) 
#0  0x000000000042ba51 in VerilatedVpiImp::cbReasonRemove(VerilatedVpioCb*) ()
#1  0x0000000000424166 in vpi_remove_cb ()
#2  0x00007fefad9ba6b1 in virtual thunk to VpiValueCbHdl::cleanup_callback() () at /usr/include/c++/10/ext/new_allocator.h:103
#3  0x00007fefad9a2dea in GpiValueCbHdl::run_callback (this=0x1145748) at cocotb/share/lib/gpi/GpiCbHdl.cpp:173
#4  0x00007fefad9b6def in handle_vpi_callback (cb_data=<optimized out>) at cocotb/share/lib/vpi/VpiImpl.cpp:583
#5  0x000000000042bf83 in VerilatedVpiImp::callValueCbs() ()
#6  0x00000000004209f3 in VerilatedVpi::callValueCbs() ()
#7  0x0000000000404a33 in main ()
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
(gdb) 
(gdb) cont
Continuing.

^C
Program received signal SIGINT, Interrupt.
0x000000000042c700 in std::_List_iterator<VerilatedVpioCb*>::operator*() const ()
(gdb) bt
#0  0x000000000042c700 in std::_List_iterator<VerilatedVpioCb*>::operator*() const ()
#1  0x000000000042ba4e in VerilatedVpiImp::cbReasonRemove(VerilatedVpioCb*) ()
#2  0x0000000000424166 in vpi_remove_cb ()
#3  0x00007fefad9ba6b1 in virtual thunk to VpiValueCbHdl::cleanup_callback() () at /usr/include/c++/10/ext/new_allocator.h:103
#4  0x00007fefad9a2dea in GpiValueCbHdl::run_callback (this=0x1145748) at cocotb/share/lib/gpi/GpiCbHdl.cpp:173
#5  0x00007fefad9b6def in handle_vpi_callback (cb_data=<optimized out>) at cocotb/share/lib/vpi/VpiImpl.cpp:583
#6  0x000000000042bf83 in VerilatedVpiImp::callValueCbs() ()
#7  0x00000000004209f3 in VerilatedVpi::callValueCbs() ()
#8  0x0000000000404a33 in main ()
(gdb) cont
Continuing.
^[[A^[[B^C
Program received signal SIGINT, Interrupt.
0x000000000042ba55 in VerilatedVpiImp::cbReasonRemove(VerilatedVpioCb*) ()
(gdb) cont
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x000000000042ba2f in VerilatedVpiImp::cbReasonRemove(VerilatedVpioCb*) ()
(gdb) cont
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x000000000042c6ca in std::_List_iterator<VerilatedVpioCb*>::operator++() ()

```

With the timer version of the debug example, the verilated design gets stuck between 

`0x000000000042c6ca in std::_List_iterator<VerilatedVpioCb*>::operator++() ()`

and 

`#0  0x000000000042ba51 in VerilatedVpiImp::cbReasonRemove(VerilatedVpioCb*) ()`

and

`0x000000000042c700 in std::_List_iterator<VerilatedVpioCb*>::operator*() const ()`

ktbarrett, 2020-10-05:
I figured that most of this logic is implemented in verilator, but it's not. It's located [in cocotb](https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/verilator/verilator.cpp). Perhaps the logic here is a bit off.

markusdd, 2020-10-06:
...which would be very sad.
But as Verilator is actually fully programmable, I would expect this can definetly be done.
But this also goes way beyond my actual knowledge, especially when it comes to VPI callbacks etc.

Anyone from verilator who could take a look at the cocotb TB interface @ktbarret linked in his post and give a recommendation on how to properly implement this? @wsynder @elgorwi @wallento ?

ktbarrett, 2020-10-06:
Excerpt from "The Verilog PLI Handbook" by Stuart Sutherland about how a simulation time step is evaluated and where the VPI callbacks fit in.

[image]
[image]

ktbarrett, 2020-10-06:
Sorry, I was not paying attention to what repo I am in! @markusdd this discussion should stay on cocotb.

garmin-mjames, 2020-10-06:
It looks like there is an issue when registering for a value change callback while inside a value change callback.

Scenario:
1. Register a value change callback for `sig1`.
2. In `sig1` callback, remove registered callback for `sig1`, do some things, then register again for `sig1` value change callback. (This is a common scenario in cocotb where you want to wait for successive rising edges of a clock, etc)
3. Return from callback.

The problem appears to be that `VerilatedVpi::callValueCbs()` iterates over everything in `s_s.m_cbObjLists[cbValueChange]`, but `VerilatedVpiImp::cbReasonAdd` adds to the end of this. Also as far as I can tell the `prevDatap` value isn't updated until after iteration occurs, so the newly registered callback is called again during the current iteration.

ktbarrett, 2020-10-06:
@garmin-mjames Is [this](https://github.com/verilator/verilator/blob/c5da072ff680dd64eca10f18bd8e97f6a1643073/include/verilated_vpi.cpp#L497) the line you are refering to? Just moving the end check into a const variable and using that would prevent that issue, right? `callCbs` and `callTimedCbs` just above it have the same bug.

garmin-mjames, 2020-10-06:
@ktbarrett yeah, that'll keep iterating on the mutated list.

PR #2588 (2020-11-06): Fix iteration over mutating list bug in VPI impl
Previously, in any given VPI callback, if the callback body registered the same callback, that callback would be processed in the currently executing call to the call*Cbs function. In the worse case, this could lead to an infinite loop. It is a bug in the class of "modifying the loop you are iterating over".

Closes #2536. Well, at least the verilator aspect of that issue; it is duplicated in the cocotb repo (cocotb/cocotb#2051).
PR #2644 (2020-11-17): Fix VPI callback list iteration
Actually fix #2536. This is a follow-on to #2588 with a test.

Saving the **end()** iterator isn't enough, as it points to a non-existent element after the last element in the list.
Callbacks added with **push_back()** are inserted before it, so the same problem still exists.

We need to get the last element in the list when the loop starts, and exit once it's been processed.
I played with keeping an exit condition in the for loop, but I couldn't solve a segfault I was getting when dealing with the last element being erased.

If you have suggestions on a differe [... cut]
closed 2020-11-06 by wsnyder