verilator/verilator #2537: clang-10 complains: warning: bitwise negation of a boolean expression; did you mean logical negation?

yurivict, 2020-09-09:
clang-10 complains about ```~``` in this code:
```
    if ((1U & (~ ((((((((1U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                         [0U][0U]) | (2U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                                      [0U][0U])) |
                       (3U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                        [0U][0U])) | (4U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                                      [0U][0U])) |
                     (5U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                      [0U][0U])) | (6U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                                    [0U][0U])) | (7U
                                                  ==
                                                  vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                                                  [0U]
                                                  [0U]))
                  | (8U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
                     [0U][0U]))))) {
```

warning:
```
../CMakeFiles/x.dir/Vsmth_smt.dir/Vsmth_smth__33.cpp:8650:16: warning: bitwise negation of a boolean expression; did you mean logical negation? [-Wbool-operation]
    if ((1U & (~ ((((((((1U == vlTOPp->smth_smth__DOT____Vcellinp__blah_blah____pinNumber32
              ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               !
```

Verilator 4.040 2020-08-15 rev v4.038-32-gbdecf6c4e
FreeBSD 12.2

wsnyder, 2020-09-10:
Verilator during configure should be detecting and passing -Wno-bool-operation.  Can you see if that was part of your verilator-config.cmake file?

yurivict, 2020-09-10:
Sorry, it was my fault, I forgot to add ```VERILATOR_CFLAGS``` to ```target_compile_options```.

Thank you!

closed 2020-09-10 by yurivict