verilator/verilator #2538: %Error: Verilator threw signal 9. Suggest trying --debug --gdbbt

ayewinoung, 2020-09-10:
How may we help - what is your question?
Hi 
I've been using verilator for this project for multiple sub modules.  When I combine the whole chip design, I do get above message and verilator do not compile the project. 
I've sub modules projects and complex verifications works well.  Only when I do complete top level chip.
To be fair its quite a big top level chip, with ~ 4096 simple CPU cores.

Any thoughts, comments, tricks to debug and get around, much appreciated.

wsnyder, 2020-09-10:
And what do you get for output when you try what it suggests (--debug --gdbbt)?

Are you sure it's not just out of memory?

You might also want to try the new hierarchical options, which should be able to partition the top logic from the cores.

ayewinoung, 2020-09-10:
Thanks, I'm trying with (--debug --gdbbt) but doesn't finish or even start the compile.
"out of memory" is good point checking it as well, and will try with 128G machine.
How do I invoke "hierarchical options" ?   Can you give me some clues on which option I should read up?

Thanks in advance.

wsnyder, 2020-09-10:
You can try "--debug --gdbbt --no-debug-leak" which will use less memory, but memory does seem the problem.

For hierarchical see the HIERARCHICAL VERILATION section of the manual.

wsnyder, 2020-09-10:
Also make sure your shell's "ulimit" is set to unlimited.

ayewinoung, 2020-09-11:
Thanks wsnyder, it is indeed running out of memory, even with 128G and taking more than 2hrs to compile.  And I tried to use hierarchical variation, but run into some snugs.
1. When I use /* verilator hier_block */ in the verilog module as specified. It come back with " Unknown verilator comment: /*verilator hier_block*/"
2. When I use " --hierachical " in ventilator flags it comes back with " %Error: Invalid option: --hierachical "

Complete compile command:
"verilator  -sc --exe --threads 8 -O2 -x-assign 0 -Wall --trace --assert --hierachical  --top-module ... "

Looks like I'm missing something here, any help appreciated.

wsnyder, 2020-09-11:
You need a newer version of Verilator, hierarchical was recently added.

Closing this out - if you have other non-trivial problems, please open a new bug so we can get you the right assistance. Thanks.

closed 2020-09-11 by wsnyder