verilator/verilator #2539: Infinite loop when compiling with bad enum/typedef syntax

codypiersall, 2020-09-10:
First, let me say **thank you** for this program.  Being able to use it is a joy!

------------------------

Verilator goes into an infinite loop whenever a syntactically invalid typedef of an enum that uses the same name for the enumeration and typedef.  Here's the example code that reproduces the issue:

```systemverilog
/* Filename is "top.sv"*/
module test();

typedef enum oops [2:0] {
  SOMETHING
} oops;

endmodule;
```

Attempting to compile (or even lint) with `verilator --lint-only top.sv` causes an infinite loop; no output, but 100% CPU utilization forever.

If the enum tag and the typedef name are not the same, verilator will correctly report an error instead of infinite looping.

I'm running the latest version of verilator from GitHub:

```console
cody@compy $ ./bin/verilator --version
Verilator 4.101 devel rev v4.100-7-gd0819f15
```

I thought this was related to #1388, but this is a distinct issue.

wsnyder, 2020-09-10:
Makes sense.  Would you be willing to try a patch?  Basically make a test case, run with "--debug --gdb" and hit ctrl-c and "bt" to see where in the code the hang is, then add protection similar to the other "Self-referential" errors in V3LinkDot.

codypiersall, 2020-09-11:
Sure!  May not be for a while though.

wsnyder, 2020-11-07:
Fixed, thanks for your suggestion.

closed 2020-11-07 by wsnyder