verilator/verilator #2540: Memory leaks

yurivict, 2020-09-11:
I've ran verilator under Google Perftools profiler, and its log ended with this:
```
...skipped...
Dumping heap profile to my-google-profile-pid22507.0094.heap (60570 MB allocated cumulatively, 1605 MB currently in use)
Dumping heap profile to my-google-profile-pid22507.0095.heap (Exiting, 1601 MB in use)
```
The last line means that 1601 MB were lost.

Here is the same "Exit" profile as pdf: https://people.freebsd.org/~yuri/verilator-memory-leaks-profile.pdf
It shows stack paths in which memory was lost in the course of the run of verilator.

If these leaks would be fixed, this might reduce the overall memory footprint and allow verilator to work on larger designs.

Version 4.040
clang-10
FreeBSD 12.2

wsnyder, 2020-09-11:
This seems inside the std library, but perhaps an ifstream isn't getting freed, can you please build with -DVL_LEAK_CHECKS, then run the _dbg binary and see what that backtrace suggests?

Note Verilator intentionally does not free certain fileline structures for performance but this won't matter in the big picture, VL_LEAK_CHECKS fixes that and also adds some internal tracking.

yurivict, 2020-09-11:
Indeed, with ```-DVL_LEAK_CHECKS``` and revision ```4.100-8-gfac04521``` the memory leak is gone. The memory profile is also significantly reduced.

But even with this option Verilator is still terribly memory-hungry. It grows to 35GB on a not-too-large design, and never finishes.

wsnyder, 2020-09-11:
Yes, Verilator always flattens the design which can use a lot of memory, 35GB isn't large compared to some ;). Try the new hierarchical compilation method, assuming you can find a good splitting point (e.g. CPU core).

closed 2020-09-11 by wsnyder