verilator/verilator #2541: Error on conditional logic inside output port of generate construction

aignacio, 2020-09-11:
Hi everyone,

I'm designing a NoC using verilator but there's a strange error when I try to build the following design below. Basically it doesn't allow me to write custom logic inside the output port of the module inside the generate construction, when I remove the conditional operator in the output fin_resp_west_o, fout_req_west_o it builds with no issues but once I put a condition to connect to a different output it fails to build. Please check the snippet below and the following changes.

```verilog
genvar x_idx,y_idx;
  generate
    for(x_idx=0;x_idx<NOC_CFG_SZ_X;x_idx++) begin : noc_lines
      for(y_idx=0;y_idx<NOC_CFG_SZ_Y;y_idx++) begin : noc_collumns
        parameter west_y_idx = y_idx-1;
        router_sample#(
          .ROUTER_X_ID(x_idx),
          .ROUTER_Y_ID(y_idx)
        ) u_router (
          .clk              (clk),
          .arst             (arst),
          // North
          .fin_req_north_i  ('0),
          .fin_resp_north_o (),
          .fout_req_north_o (),
          .fout_resp_north_i('0),
          // South
          .fin_req_south_i  ('0),
          .fin_resp_south_o (),
          .fout_req_south_o (),
          .fout_resp_south_i('0),
          // West
          .fin_req_west_i   (y_idx>0?ew_req[x_idx][west_y_idx]:'0),
          .fin_resp_west_o  (y_idx>0?ew_resp[x_idx][west_y_idx]:corner_resp_left[x_idx][y_idx]),
          .fout_req_west_o  (y_idx>0?we_req[x_idx][west_y_idx]:corner_req_left[x_idx][y_idx]),
          .fout_resp_west_i (y_idx>0?we_resp[x_idx][west_y_idx]:'0),
          // East
          .fin_req_east_i   (y_idx<(NOC_CFG_SZ_Y-1)?we_req[x_idx][y_idx]:'0),
          .fin_resp_east_o  (we_resp[x_idx][y_idx]),
          .fout_req_east_o  (ew_req[x_idx][y_idx]),
          .fout_resp_east_i (y_idx<(NOC_CFG_SZ_Y-1)?ew_resp[x_idx][y_idx]:'0),
          // Local
          .local_mosi_i     (local_mosi[x_idx][y_idx]),
          .local_miso_o     (local_miso[x_idx][y_idx])
        );
      end
    end
  endgenerate
```
The code above gives the following output:
```bash
%Error-UNSUPPORTED: src/noc.sv:51:7: Unsupported: Can't unroll generate for; genvar assigned *inside* loop
                                       : ... In instance noc
   51 |       for(y_idx=0;y_idx<NOC_CFG_SZ_Y;y_idx++) begin : noc_collumns
      |       ^~~
%Error: src/noc.sv:51:7: For loop doesn't have genvar index, or is malformed
                           : ... In instance noc
   51 |       for(y_idx=0;y_idx<NOC_CFG_SZ_Y;y_idx++) begin : noc_collumns
      |       ^~~
%Error: Internal Error: src/noc.sv:51:7: ../V3Param.cpp:517: GENFOR should have been wrapped in BEGIN
                                           : ... In instance noc
   51 |       for(y_idx=0;y_idx<NOC_CFG_SZ_Y;y_idx++) begin : noc_collumns
      |       ^~~
                        ... See the manual and https://verilator.org for more assistance.
make: *** [output/Vnoc.mk] Error 1
```
If I modify just for test (it's not the correct connection) the following outputs it builds with no issues:
```verilog
...
.fin_resp_west_o  (ew_resp[x_idx][y_idx]),
.fout_req_west_o  (we_req[x_idx][y_idx]),
...
```

**Verilator version =** Verilator 4.040 2020-08-15 rev v4.038-32-gbdecf6c4e
**Tried with the master version**
**Build command =** 
*verilator -CFLAGS " -g3 -Wall -Werror -DWAVEFORM=\"/tmp/ravenoc.fst\" -DEN_TRACE=\"1\"" --top-module ravenoc --Mdir output --Wno-UNOPTFLAT -O3 --exe --threads  4 --trace --clk                 clk --trace-fst --trace-structs --trace-threads 4 --trace-underscore --trace-depth                      10000 --trace-max-array 10000 --trace-max-width 10000 --cc incdir+src/include +define+SIMULATION src/include/ravenoc_defines.svh src/include/ravenoc_structs.svh src/include/ravenoc_pkg.sv  src/fifo.sv src/output_module.sv src/ravenoc.sv src/include/ravenoc_pkg.sv src/router_ravenoc.sv src/rr_arbiter.sv src/vc_buffer.sv src/input_router.sv src/input_module.sv src/input_datapath.sv tb/testbench.cpp -o ravenoc*

wsnyder, 2020-09-11:
I speculate what it should really be giving an UNSUPPORTED error on is this:

        parameter west_y_idx = y_idx-1;

Can you please replace west_y_idx with "y_idx-1" in your cell and see if that works?

wsnyder, 2020-09-11:
Ok, not sure what is happening. Can you please attach a self contained minimized example that shows the error ideally in test_regress format?

aignacio, 2020-09-11:
Thanks for answer @wsnyder, could you please point me what's the test_regress format? Anyway the simple example below shows the same problem, apparently you **cannot have conditions inside an output** of a generated module.
 
Example tested:
```verilog
 module tb();
  logic [2:0] [2:0] in;
  logic [2:0] [2:0] out;

  genvar x,y;
  generate
    for(x=0;x<3;x++) begin
      for(y=0;y<3;y++) begin
        test u_test(
          .in   (in[x][y]),
          .out  (y>0?out[x][y-1]:out[x][y])
        );
      end
    end
  endgenerate
endmodule

module test(
  input   in,
  output  out
);
  assign out = in;
endmodule
```

Verilator output:
```bash
%Error-UNSUPPORTED: src/tb.sv:54:7: Unsupported: Can't unroll generate for; genvar assigned *inside* loop
                                       : ... In instance tb
   54 |       for(y=0;y<3;y++) begin
      |       ^~~
%Error: src/tb.sv:54:7: For loop doesn't have genvar index, or is malformed
                           : ... In instance tb
   54 |       for(y=0;y<3;y++) begin
      |       ^~~
%Error: Internal Error: src/tb.sv:54:7: ../V3Param.cpp:517: GENFOR should have been wrapped in BEGIN
                                           : ... In instance tb
   54 |       for(y=0;y<3;y++) begin
      |       ^~~
                        ... See the manual and https://verilator.org for more assistance.
make: *** [output/Vravenoc.mk] Error 1
```

aignacio, 2020-09-11:
Tested here, same happens with a single generate loop...
```verilog
 module tb();
  logic [2:0] in;
  logic [2:0] out;

  genvar x;
  generate
      for(x=0;x<3;x++) begin
        test u_test(
          .in   (in[x]),
          .out  (x>0?out[x]:out[x])
        );
    end
  endgenerate
endmodule

module test(
  input   in,
  output  out
);
  assign out = in;
endmodule
```

wsnyder, 2020-09-15:
test_regress format is described in docs/internal.adoc.  I think I see what's going on basically it thinks the output is assigning to the genvar which it isn't, if you could make the test self checking (again see the docs and/or test_regress/EXAMPLE.pl) then I think we'll be close.

wsnyder, 2020-09-15:
Looking into fixing this... this code is actually illegal.  An output port must connect to a wire, it's not legal to conditionally select.  So this is really something lint should error out on. For fixing the linting the provided test is sufficient.

aignacio, 2020-09-20:
Thanks for the heads up but if you see through it's not connecting the output to a mux, it's connecting to a wire, the thing is that by the generate in the elab time it'll select which datapath it should connect...