verilator/verilator #2544: cmake build fails when --hierarchical option is used

yurivict, 2020-09-11:
With the ```--hierarchical``` option the configuration fails:
```
-- Executing Verilator...
CMake Error at verilog/CMakeFiles/verilog-simulator-library.dir/Vtest_bench.dir/Vtest_bench_copy.cmake:47 (target_link_libraries):
  Cannot specify link libraries for target "Vtest_bench" which is not built
  by this project.
Call Stack (most recent call first):
  /usr/local/share/verilator/verilator-config.cmake:218 (include)
  verilog/CMakeLists.txt:61 (verilate)
```

Specifically, the file ```Vtest_bench_copy.cmake``` has a new section in the end:
```
# Verilate hierarchical blocks
add_library(Vcore_1 STATIC)
target_link_libraries(Vtest_bench PRIVATE Vcore_1)
verilate(Vcore_1 PREFIX Vcore_1 TOP_MODULE core_1 DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/Vcore_1 SOURCES  /usr/home/yuri/x/verilog/test_bench.v VERILATOR_ARGS -f /usr/home/yuri/x/verilog/CMakeFiles/x.dir/Vtest_bench.dir/Vcore_1_hierCMakeArgs.f -CFLAGS -fPIC)
...
```

The second line in this section causes the above error.

What might be wrong?

yurivict, 2020-09-15:
Here is a basic testcase using the ```--hierarchical``` option that breaks: https://people.freebsd.org/~yuri/verilator-hierarchical-testcase.tgz

```
cmake .
<...skipped...>
-- Executing Verilator...
CMake Error at CMakeFiles/verilog-top-target.dir/Vtest_bench.dir/Vtest_bench_copy.cmake:47 (target_link_libraries):
  Cannot specify link libraries for target "Vtest_bench" which is not built
  by this project.
Call Stack (most recent call first):
  /usr/local/share/verilator/verilator-config.cmake:218 (include)
  CMakeLists.txt:10 (verilate)
```

@wsnyder It never creates the cmake target ```Vtest_bench``` but it assumes that this target exists. Do you know what might be wrong? The target name ```Vtest_bench``` is derived from the ```SOUCRES``` supplied to the ```verilate()``` function.

The offending line is printed by verilator here: https://github.com/verilator/verilator/blob/master/src/V3EmitCMake.cpp#L214

yTakatsukasa, 2020-09-15:
The failure happens on my side too. I will take a look at this trouble.

yurivict, 2020-09-16:
The same testcase also breaks with the ```gmake``` method:
```
$ verilator  --cc --hierarchical test_bench.v my_tile.v 
gmake: Entering directory '/usr/home/yuri/x/verilator-hierarchical-testcase/obj_dir'
gmake[1]: Entering directory '/usr/home/yuri/x/verilator-hierarchical-testcase'
/usr/local/bin/verilator -f obj_dir/Vmy_tile_hierMkArgs.f
gmake[1]: Leaving directory '/usr/home/yuri/x/verilator-hierarchical-testcase'
gmake[1]: Entering directory '/usr/home/yuri/x/verilator-hierarchical-testcase'
/usr/local/bin/verilator -f obj_dir/Vtest_bench_hierMkArgs.f
%Error-TIMESCALEMOD: obj_dir/Vmy_tile/my_tile.sv:6:8: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     test_bench.v:3:8: ... Location of module with timescale
    3 | module test_bench (clk);
      |        ^~~~~~~~~~
%Error: Exiting due to 1 error(s)
        ... See the manual and https://verilator.org for more assistance.
gmake[1]: *** [obj_dir/Vtest_bench_hier.mk:30: hier_launch_verilator] Error 1
gmake[1]: Leaving directory '/usr/home/yuri/x/verilator-hierarchical-testcase'
gmake: *** [Vtest_bench_hier.mk:34: Vtest_bench.mk] Error 2
gmake: Leaving directory '/usr/home/yuri/x/verilator-hierarchical-testcase/obj_dir'
%Error: gmake -C obj_dir -f Vtest_bench_hier.mk   hier_verilation exitted with 2
%Error: Command Failed /usr/local/bin/verilator_bin --cc --hierarchical test_bench.v my_tile.v
```

It complains that timescale isn't specified when it is specified.

yTakatsukasa, 2020-09-16:
Several independent issues are found. I will make a few more PRs.

yTakatsukasa, 2020-09-19:
#2560 should fix the issue (at least on my local computer).
Thanks for the report.

PR #2560 (2020-09-19): Fix cmake build with --hierarchical option
Fixes #2544 

There were some problems.
1. .cmake generated by `hier_block` assumed the target name is `--prefix`.
2. `DIRECTORY` option was not properly passed via `-Mdir` option

To test the item 1. above,  simple dedicated CMakeLists.txt is used in 5d99340cf022a1225ae14c093f044b9924821b4f .
closed 2020-09-19 by yTakatsukasa