verilator/verilator #2545: Help on using "--hierarchical"

ayewinoung, 2020-09-12:
How may we help - what is your question?
Looking to try out "--hierarchical" for big design and got error when I set "--hierarchical" flag at compile time.

%Error: Specified --top-module 'example_module' was not found in design.
%Error: Exiting due to 1 error(s)

'example_module' is *not* top level of the chip design but sub module with "/* verilator hier_block */" inside the module.  It compile and runs fine, without "--hierarchica" flag, but I suspect its not doing any hierarchial anyway.

Verilator 4.100 2020-09-07 rev v4.040-74-g16fba5948

Any help much appreciated.

yTakatsukasa, 2020-09-12:
As the hierarchical Verilation internally calls child verilator multiple times for each `hier_block`, `example_module` is set to top module in one of the child run.

Can you provide minimal self-contained example with full command line option ? That is the easiest way to see the same page.

Here is some hints to make minimal examples.
I guess `example_module` can be almost empty.
Other modules that are not marked `hier_block` can be eliminated.
Ports can be minimal.

I am curious especially about the following things.
- Is `-v` or `-y`  used ?
- Is `example_module` is parameterized ?
- Are modules in the same file ?  or in different files ?

ayewinoung, 2020-09-12:
Hi Thanks for the tips, its helpful.  I'll write up my design "hierarchy" and share here, its bit complicated.  And I'll try to describe in words here and answer your questions:

1.  I don't have either "-v" or "-y", may be should I use them ? exact command from execution: 
    verilator  -sc --exe --threads 8 -O2 -x-assign 0 -Wall --top-module sirius_tb --prefix sirius_tb  <bunch of systemC, C, verilog files>
2. "example_module" is *not* parameterized, but there are other files in the design that are parameterized.
3. "example_module" is within another "hier_block" module, "example_module" is leaf so to speak.  I'll try with just one "hier_block" module approach and report back.
4. All the modules are in different files, so one file, one module so to speak.
5. One interesting thing to mention is, "example_module" has two instances of the same module, call it "module_a".  "module_a" itself instantiate two other different modules, call it "module_b" and "module_c".  None of the module "a, b, & c" has any parameterized values.  They do however, use share "function" which is implemented in "include verilog file".
6. One other thing also may be of interest is, "example_module" is instantiated x65 times in "module_d" using verilog generate statement with genvar
7. Just to complete the picture,
    There is  "module_e" which instantiate "module_d" x4 times.  2x2 Tile
    There is "module_f" which instantiate "module_e" x4 times.   4x4 Tile
    There is "module_g" which instantiate "module_f" x4 times.  8x8 Tile
    There is "module_h" which instantiate "module_g" x4 times. 16x16 Tile
    There is "module_i" which instantiate "module_h" x4 times. 32x32 Tile
    There is "module_u" which instantiate "module_i" x4 times. 64x64Tile
    Finally there is top level module which instantiate "module_u" and some other modules.
8. There are other modules in design which are not part of this "hierarchy", so to speak, but are parametrizable.

Feel free to let me know if you need any more details in specific areas.  Unfortunately, I can't share the RTL.

ayewinoung, 2020-09-12:
One more update, I try to experiment with just one module with "/* verilator hier_block */" and start with most simplest module , in above example "module_b", but still fill with same error message.

It seems that, the tool is not finding the verilog file for the module with "/* verilator hier_block */", may be something to do with the way I pass the file names ? I pass all the files with explicit path and all in command line.

I'll try with "-y"

yTakatsukasa, 2020-09-12:
Hierarchical verilation creates protect-lib for each hier_block. A user moudle of hier_block reads the wrapper of protect-lib instead of the original design.
It seems that the one of the user module can not read the generated wrapper.
Tests in the test suite does not use `-v`, so it should be fine.

When you run the verilator with --hierarchical option, you may see the message like below.
```]
make[1]: Entering directory '/mnt/prj/verilator/test_regress'
/mnt/prj/verilator/bin/verilator -f obj_vlt/t_hier_block/Vdelay_2_hierMkArgs.f
make[1]: Leaving directory '/mnt/prj/verilator/test_regress'
make[1]: Entering directory '/mnt/prj/verilator/test_regress'
/mnt/prj/verilator/bin/verilator -f obj_vlt/t_hier_block/Vdelay_9_hierMkArgs.f
make[1]: Leaving directory '/mnt/prj/verilator/test_regress'
make[1]: Entering directory '/mnt/prj/verilator/test_regress'
/mnt/prj/verilator/bin/verilator -f obj_vlt/t_hier_block/Vt_hier_block_hierMkArgs.f
make[1]: Leaving directory '/mnt/prj/verilator/test_regress'
```

Can I see the .f file of the verilator run when you hit the error message.

I believe the logic of the design is not related to this error. You can remove any logics, signals from the design.
The hierarchy and how the modules are instantiated (and maybe name or directory structure) matter.
Without the test case, I'm afraid it will take so long discussion.

ayewinoung, 2020-09-13:
Some good news.. I can make "-- hierarchical" to most atomic verilog module, just to prove out Makefile and process, by using "-v /complete/path/to/verilog.v" 

However, bad news is when I change to module up a level, where I really want to do hierarchical compile, I got different error.

%Warning-UNOPTTHREADS: Thread scheduler is unable to provide requested parallelism; suggest asking for fewer threads.
                       ... Use "/* verilator lint_off UNOPTTHREADS */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)

I've attached both, log out put of compile, as well as .f file and .mk files generated by tools.  Had to change file extension to .txt to update it here.

sirius_tb : Top level of RTL module
bc_hash_engine : is the RTL module with /* verilator hier_block */

FYI - I'm running on 16 cores/32 threads machine with 128GByte of memory and nothing else is running, in Ubuntu 18.04

[compile.log](https://github.com/verilator/verilator/files/5213548/compile.log)

[sirius_tb_classes.mk.txt](https://github.com/verilator/verilator/files/5213549/sirius_tb_classes.mk.txt)

[sirius_tb_hier.mk.txt](https://github.com/verilator/verilator/files/5213550/sirius_tb_hier.mk.txt)
[sirius_tb_hierMkArgs.f.txt](https://github.com/verilator/verilator/files/5213551/sirius_tb_hierMkArgs.f.txt)
[sirius_tb.mk.txt](https://github.com/verilator/verilator/files/5213552/sirius_tb.mk.txt)
[Vbc_hash_engine_hierMkArgs.f.txt](https://github.com/verilator/verilator/files/5213553/Vbc_hash_engine_hierMkArgs.f.txt)

yTakatsukasa, 2020-09-13:
Thanks, I will look in to log files.

Even with UNOPTTHREADS  warning, the simulated result is correct. You can ignore the warning by adding `-Wno-fatal`.

ayewinoung, 2020-09-13:
All good, I think I resolve my issues to get the hierarchical compile going and running the project.

Thanks for taking the time to provide the hints/tips & guides.

Will report back with comparison of compile time vs run-time vs memory foot print, for hierarchal vs flat compile and runs, for reference.

closed 2020-09-13 by ayewinoung