verilator/verilator #2546: Systemverilog static casting issue V4.100

xver, 2020-09-13:
The following code worked fine on 4.38, but from 4.40 it starts to fail with:

%Warning-WIDTH: ../src/Target.sv:22:82: Operator ADD expects 16 bits on the LHS, but LHS's VARREF 'shortV_i' generates 32 bits.
                                      : ... In instance Target
   22 |       foreach(ShortV_exp[shortV_i]) ShortV_exp[shortV_i] = 100+shortint'(shortV_i+1);
      |                                                                                  ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.

--------
/* verilator lint_off VARHIDDEN */
 /* verilator lint_off UNUSED */
module automatic Target(input reg clk_i);   
   int 	        shortV_i;
   shortint     ShortV_exp[4]; 
   initial begin
      foreach(ShortV_exp[shortV_i]) ShortV_exp[shortV_i] = 100+shortint'(shortV_i+1);
      foreach(ShortV_exp[shortV_i])    $display("ShortV_exp[%0d]=%0h",shortV_i,ShortV_exp[shortV_i]);   
end    
endmodule : Target

Thanks Victor

wsnyder, 2020-09-15:
I think the warning is correct.  100 is a 32-bit number, so if you have an overflow you'll lose it.  I think your code should be

    foreach(ShortV_exp[shortV_i]) ShortV_exp[shortV_i] = shortint'(100 + shortV_i+1);

Agreed?

xver, 2020-09-16:
Agree , but I checked it with a "big 3" sim  , and even most orthodox one (Questa) is not complaining  about it.  
Thanks Victor

wsnyder, 2020-09-16:
The big 3 simulator don't have lint warnings AFAIK, you need to use a lint tool.

closed 2020-09-16 by wsnyder

xver, 2020-09-16:
agree it pass even with no casting.