verilator/verilator #2547: Compiler error message with SystemVerilog stream operator

xver, 2020-09-13:
I get a compile error for the following code:

SV : 
------------------------------------------
/* verilator lint_off VARHIDDEN */
 /* verilator lint_off UNUSED */
/* verilator lint_off WIDTH */
module automatic Target(input reg clk_i);
   
   logic [31:0]         mem_data; 
   byte                 data_in[4];
   
   initial begin 
      foreach(data_in[i])  data_in[i] = 'h10*(i+1);
      mem_data =  {>>{data_in}};
      $display("TEST: mem_data=%0h",mem_data );	
   end
   
endmodule : Target
/* verilator lint_on VARHIDDEN */
/* verilator lint_on UNUSED */
/* verilator lint_on WIDTH */
 
Error
----------------
in file included from Vcs_common__ALL.cpp:8:0:
Vcs_common__Slow.cpp: In static member function ‘static void Vcs_common::_initial__TOP__1(Vcs_common__Syms*)’:
Vcs_common__Slow.cpp:42:82: error: cast from ‘CData* {aka unsigned char*}’ to ‘IData {aka unsigned int}’ loses precision [-fpermissive]
     vlTOPp->Target__DOT__mem_data = (0xffU & (IData)(vlTOPp->Target__DOT__data_in));
                                                                                  ^

wsnyder, 2020-09-15:
I think this is an unsupported case that it doesn't warn about.  Perhaps you would be willing to submit a self-checking test (see docs/internals.adoc for the details), and ideally work on improving this support?

wsnyder, 2023-09-26:
Recent version gives

`%Error-UNSUPPORTED: t/t_0.v:33:20: Unsupported: Stream operation on a variable of a type 'byte$[0:3]'
`

wsnyder, 2024-09-14:
This works now, probably with #5326.

closed 2024-09-14 by wsnyder