verilator/verilator #2548: SystemVerilog stream operator non-datatype error

xver, 2020-09-13:
yet another stream operator issue : 

Error
--------
%Error: ../src/Target.sv:26:10: Operator STREAML expected non-datatype RHS but 'byte' is a datatype.
                              : ... In instance Target
   26 |       {<<byte{data_out}} = mem_data;
      |          ^~~~
%Error: Exiting due to 1 error(s)

SV : 
---------
module automatic Target(input reg clk_i);
  
   logic [31:0]         mem_data; 
   byte 		data_in[4];
   byte 		data_out[4];
   
   initial begin 
      foreach(data_in[i])  data_in[i] = i+1;
      mem_data =  {<<8{data_in}};
      $display("TEST: mem_data=%h",mem_data );
      {<<byte{data_out}} = mem_data;
      foreach(data_out[i])  $display("TEST: data_out[%0d]=%h",i,data_out[i]);
   end

endmodule

wsnyder, 2020-09-15:
Like #2547, I think this is an unsupported case that it doesn't warn about.  Perhaps you would be willing to submit a self-checking test (see docs/internals.adoc for the details), and ideally work on improving this support?

xver, 2020-09-16:
Ok , let me try . Thanks Victor

xver, 2020-09-27:
Wilson, Could you review the attached zip file (test+pl).  I am not quite sure, that I fulfilled all requirements...
[t_stream_integer_type_bad.zip](https://github.com/verilator/verilator/files/5287540/t_stream_integer_type_bad.zip)

wsnyder, 2020-09-27:
Yes, that's the right idea and seems very thorough, just a few minor things I can fix.  It passes on one commercial simulator but fails on another. (below)

If you could make this a pull request that's a bit better but I can deal with the ZIP.  I do however need a pull request from you adding your name to docs/CONTRIBUTORS, or indication in this issue that I can do so for you and you recognize the contribution as described there, so it's legally kosher.  Thanks.

```
*-* TEST failed error integer_atom_type time *-*:
time_in[0]=501, time_out=xxxxxxxxxxxxxxxx
time_in[1]=502, time_out=xxxxxxxxxxxxxxxx
time_in[2]=503, time_out=xxxxxxxxxxxxxxxx
time_in[3]=504, time_out=xxxxxxxxxxxxxxxx
packed_time_256=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx,packed_time_256=5040000000000000503
```

xver, 2020-09-27:
I'll check this test against commercial simulators once again and I'll do a pull request tomorrow. Also, if this issue is not urgent, I would like to try to make a patch as well. Thanks, Victor

xver, 2020-09-27:
Just to confirm. The test passes 3 out of 4 commercial simulators. 
IMHO, to pack and unpack the "time" type is not quite practical. 
But LRM defines  integer_atom_type as a integer_atom_type ::= byte | shortint | int | longint | integer | time
I think LRM should win ...

wsnyder, 2020-09-27:
For the next step, I think you need to change V3Width.cpp to understand these constructs probably in the AstStreamL visitor.

xver, 2020-09-28:
about the next step ... I'll start today. But I am not sure how long it may take yet.

xver, 2020-10-18:
I just finished a draft framework for the first part of the stream support. It contains only the "packs" operation for the  integer_atom_type, such as: <packed_data_32    = {<<byte or 8 {byte_in}}> and so on.
 Wilson, could you review the following changes and let me know if this is OK.
(please see attached files for details):
1.  V3Width.cpp
    - to remove Unpacked array specifics  after  if (VN_IS(nodep, UnpackArrayDType))  
   - disable   " expected non-datatype " error for stream operator "
          if (VN_IS(underp, NodeDType) && !VN_IS(nodep,StreamL) ) {  // Note the node itself, not node's data type

2.V3EmitC.cpp  visit(AstStreamL* nodep) override-add  UnpackArrayDType support and emit four functions:          
         VL_STREAML_UA_BYTE_XXX         
         VL_STREAML_UA_SHORTINT_XXX      
          VL_STREAML_UA_INT_XXX
          VL_STREAML_UA_LONGINT_XXX3. 
4. verilated.hadd integer_atom_type functions:   VL_STREAML_UA_BYTE_III - I have implementation example for this one only
   VL_STREAML_UA_BYTE_III
   VL_STREAML_UA_SHORTINT_QQI   
   VL_STREAML_UA_INT_WWI   
   VL_STREAML_UA_LONGINT_WWI
Thanks Victor
[verilator_debug.zip](https://github.com/verilator/verilator/files/5398391/verilator_debug.zip)

wsnyder, 2020-10-19:
Can you please make a pull request?  Then it's much easier to review and the tests will run automatically.  (Please search online for tutorials on making one.)

xver, 2020-10-19:
It is not ready for release,.  Anyway , I'll do pull request today .

wsnyder, 2020-10-19:
Understood, BTW you can mark a pull request as a "draft" not for merging.

goekce, 2022-11-02:
I also stumbled on this streaming operator issue today while changing the endianness of a data stream.

@xver are you going to continue your draft?

xver, 2022-11-02:
Sorry don't have a bandwidth.

wsnyder, 2025-09-08:
Passes on master, have been recent improvements in stream handling. Please open a new issue if still problems.

closed 2025-09-08 by wsnyder