verilator/verilator #2551: Getting error while running Example - hello_make_c

ShraddhaDevaiya, 2020-09-15:
Hello, I am trying to run the example of **hello_make_c** to get started.
But while running ```make``` command it is giving an error like following : 
```
-- Verilator hello-world simple example
-- VERILATE & BUILD --------
verilator -cc --exe --build -j top.v sim_main.cpp
%Error: Invalid Option: --build
make: *** [Makefile:38: default] Error 1
````
verilator version : 4.028 2020-02-06 rev v4.026-92-g890cecc1

I have checked documentation, but there is not any option mentioned like ```--build```. How can I solve this? Can anyone please help ?

wsnyder, 2020-09-15:
The documentation you are looking at (probably on the web) is for the most recent version.  Either download the most recent version, or look at the documentation for the version you are using, which will say to run "make" on the verilator output.

closed 2020-09-15 by wsnyder

ShraddhaDevaiya, 2020-09-15:
Yeah, it is working for verilator version 4.100. Thanks !