verilator/verilator #2552: Comercial enchancements contract

ayewinoung, 2020-09-15:
Hi
I'm looking for specific enhancements for verilator as commercial support and make available to community, (i.e. I'll pay someone(s) for enhancements to the tool) in areas of 1) compile time & memory footprint optimizations, "hierarchical vs flat" 2) runtime optimizations, "speed" 3) multiprocessor/thread utilization/efficiency improvements.
If anyone(s) is interested or know of who/which organization(s) might be, please reply with contact details and we can explore more on technical feasibility, schedule, deliverables and commercial terms. 

Thanks

wsnyder, 2020-09-15:
Always good to hear of requests for commercial support. Note you may also want to contact those mentioned under professional support here: https://verilator.org/verilator_commercial_support

wsnyder, 2020-10-18:
DId you find someone to handle this - please let me know privately at wsnyder@wsnyder.org.  Thanks.  I'm now deleting this comment.

wsnyder, 2020-12-13:
Hopefully you found someone, but suspect we don't need to leave this open...

closed 2020-12-13 by wsnyder