verilator/verilator #2557: Error when using internal generated clock.

yanx21, 2020-09-17:
I found a mismatch between the waveforms and my RTL codes when I  using a generated clock inside a block triggered by the same clock.
After debuging, I build the following concise testcase that can repeat my case.

testcase:
```v
// demo.v
module top (
    input   wire    clk,
    input   wire    gate,
    output  reg     f2
);
wire    clk1;
reg     f0;
reg     f1;
assign clk1 = clk & (~gate);
always @ (posedge clk) begin
    f0 <= ~f0;
    f1 <= f0;
    $display("{f0, f1, f2} == {%d, %d, %d}", f0, f1, f2);
end
always @ (posedge clk1) begin
    if (clk1) begin
        f2 <= f1;
    end
end
endmodule
```
verilator arguments are `--cc`
And here is my c++ wrapper
```cpp
// sim_main.cpp
#include "Vdemo.h"
#include "verilated.h"

vluint64_t main_time = 0;
double sc_time_stamp() { return main_time; }

int main(int argc, char** argv, char** env) {
    Verilated::commandArgs(argc, argv);
    Vdemo* top = new Vdemo;
    top->gate = 0;
    while (!Verilated::gotFinish() && main_time < 40) {
        top->eval();
        main_time++;
        top->clk = !top->clk;
    }
    delete top;
    exit(0);
}
```
The output is clear to see the problem, where `f1` and `f2` flip simultaneously. But in my design, `f2` should delayed by `f1`
```
// screen output
{f0, f1, f2} == {0, 0, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 1}
{f0, f1, f2} == {1, 0, 0}
```

In my real project, codes like this were widely exist  and more complicated, I hope Verilator can do better in this situation. If you can't get what I mean, please let me know, I will explain more.

Looking forward to your reply.

wsnyder, 2020-09-17:
This is a clock gater:

assign clk1 = clk & (~gate);

So "gate" needs a verilator clock_enable annotation, please see the documentation.

yanx21, 2020-09-17:
> This is a clock gater:
> 
> assign clk1 = clk & (~gate);
> 
> So "gate" needs a verilator clock_enable annotation, please see the [verilator documentation](https://verilator.org/verilator_doc.html).

Thanks your reply.
As you say, I tried adding clock_enable annotation.

1. mark  `clock_enable` on the `gate`
```v
module top (
    input   wire    clk,
    input   wire    gate/*verilator clock_enable*/,
    output  reg     f2
);
// ... Omit the rest
```
The result were same.  **BAD**

2. mark `clock_enable` on the `clk`
```v
module top (
    input   wire    clk/*verilator clock_enable*/,
    input   wire    gate,
    output  reg     f2
);
// ... Omit the rest
```
The result were same. **BAD**

3. mark `clock_enable` on the `clk` and the `gate`
```v
module top (
    input   wire    clk/*verilator clock_enable*/,
    input   wire    gate/*verilator clock_enable*/,
    output  reg     f2
);
// ... Omit the rest
```
The result were different, but still out of the expetation, where `f2` was all zero. **BAD**
```
// screen output
{f0, f1, f2} == {0, 0, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
{f0, f1, f2} == {0, 1, 0}
{f0, f1, f2} == {1, 0, 0}
```

4. Add a argument `--clk clk` or `--clk gate` before translating
It reported a error and stoped. **BAD**
```
%Error: Circular logic when ordering code (non-cutable edge loop)
        demo.v:20:1:      Example path: ALWAYS
        demo.v:8:9:      Example path: top.clk1
        demo.v:12:13:      Example path: ASSIGNW
        demo.v:8:9:      Example path: top.clk1
        demo.v:20:1:      Example path: ACTIVE
        demo.v:20:1:      Example path: ALWAYS
        demo.v:20:1:      Example path: ALWAYS
        demo.v:8:9:      Example path: top.clk1
        demo.v:12:13:      Example path: ASSIGNW
        demo.v:8:9:      Example path: top.clk1
        demo.v:20:1:      Example path: ACTIVE
        demo.v:20:1:      Example path: ALWAYS
%Error: Internal Error: ../V3Graph.cpp:267: Loops detected in graph:   VERTEX=0x2814960\n ALWAYS r8 c4
```

I tried 4 testcase, but none of them solved my problem.
Do you have any other solution?

Thanks again.

wsnyder, 2020-09-17:
Thanks for experimenting.

It seems to be confused by this flop/latchish thingie.  Not sure what this is supposed to represent hardware-wise.  Is using clock for both the sensitivity list and internally intentional?

```
always @ (posedge clk1) begin
    if (clk1) begin
        f2 <= f1;
    end
end
```

yanx21, 2020-09-28:
> Thanks for experimenting.
>
> It seems to be confused by this flop/latchish thingie. Not sure what this is supposed to represent hardware-wise. Is using clock for both the sensitivity list and [...]

emm... It doesn't make much sense indeed. I found the code in the project wrote by someone else, I couldn't find out the real meaning behind the code. But it's legal after all, so I think it's still a problem

markusdd, 2020-10-11:
Just to throw another opinion in here: I consider this code to be bogus and not something that must be supported.

It's not a latch, because than it would need to read always @(*) or always_latch, nor is it a proper flop, because then the level check if(clk1) is bogus.

It is essentially inserting the clock into the datapath while triggering on that exakt same clock. This is also very problematic from a synthesis perspective, because how would a tool balance setup/hold properly here?

Tomorrow I could briefly check with some commercial tooling, but I suspect that this would at least raise some critical warnings.

zenghan2088, 2021-01-11:
Aha! I got similar complaints from the Hardware team that the waveform of VCS is different from our SystemC models generated by Verilator.

I'm happy to see this STAR as I'm not the only one who troubled by this problem, and thanks for your demo, @yanx21

In short, there are two level clock gating in our RTL design, original_clk --> gated_clk_1 --> gated_clk_2, when gated_clk_2 trying to sample a signal, say d1, from gated_clk_1 clock domain to d2, d2 always gets current updated signal value of d1 instead of the previous value.

Here are what I found after a hard debuging to the verilator-generated CPP source. 
In _eval(vlSymsp), signal din is updated by sc_module_top.gate_clk_1 and sampled by __VinpClk__TOP__sc_module_top.gate_clk_2, as we know, eval_step() is called at the posdege of original_clk and _eval() is called more than one time untill stable, following are the simplified process:
__VclockLoop ==0: 
       1) (__VinpClk__TOP__sc_module_top.gate_clk_2 & ~ __Vclklast__TOP____VinpClk__TOP__sc_module_top.gate_clk_2) is false, d2 = d1 is not executed.
       2)  (sc_module_top.gate_clk_1 & ~ __Vclklast__TOP___sc_module_top.gate_clk_1) is true, d1 is updated to new_d1
       3)   __Vclklast__TOP____VinpClk__TOP__sc_module_top.gate_clk_2 = __VinpClk__TOP__sc_module_top.gate_clk_2; 
             __VinpClk__TOP__sc_module_top.gate_clk_2 = GATE_PROCESS(sc_module_top.gate_clk_1); 
            Now __VinpClk__TOP__sc_module_top.gate_clk_2 is 1
__VclockLoop ==1: 
        1) (__VinpClk__TOP__sc_module_top.gate_clk_2 & ~ __Vclklast__TOP____VinpClk__TOP__sc_module_top.gate_clk_2) is true, d2 = d1 is executed now, but it's too late, the value of d1 is new_d1 at this moment.

I think the RTL has been proven to work, it's likely a issue with Verilator.

BTW, I have tried both /*verilator clocker*/ and /*verilator clocker_enable*/ directives, they does not seem to work.

wsnyder, 2021-01-11:
Is original_clk a top level input to the design?  Assuming so, an the clocker/clock_enables are working properly then the VclockLoop should not be there - if it is it would not be surprising that there are races or other clocking problems. 

So either something is mismarked, or Verilator has a bug marking/propagation of the clocker or clock_enable attributes.

Double check that you have appropriate clock_enable attribute on what gates gate_clk_1.

If you turn on --debugi-V3Order 9, make a logfile, it will also indicate what it's tracing as a clock tree.  You can also look at the .tree file to see what has a clock_enable attribute ("[aCLKEN]" on the correct VARiables).

zenghan2088, 2021-01-13:
Yes, original_clk means the top-level input. 

I turned on "--debugi-V3Order 9" and grepped [aCLKEN] in the log file, turns out only the latch variable in the first instance of  clock_gating module is marked with [aCLKEN], _sc_topmodule__DOT__u_clkgate__DOT__latch [aCLKEN]_, but the second one is not marked, interesting.

I was desperate, so I made the following example Verilog:
module sc_example(
    input clk
  , input s_clk
  , input rst_n
  , input s_mode
  , input clk_en_1
  , input clk_en_2
  , input din
  , output dout
);

wire gated_clk_1;
wire gated_clk_2;
reg din_1;
reg din_2;

clkgate u_clk_gate_1(
  .clk_in(clk),
  .en(clk_en_1),
  .clk_out(gated_clk_1)
);

clkgate u_clk_gate_2(
  .clk_in(gated_clk_1),
  .en(clk_en_2),
  .clk_out(gated_clk_2)
);

always @ (posedge gated_clk_1 or negedge rst_n)
begin
  if (rst_n == 0)
    din_1 <= 1'b0;
  else
    din_1 <= din;
end

wire final_clk = (s_mode == 1'b0) ? gated_clk_2 : s_clk;

//always @ (posedge gated_clk_2 or negedge rst_n)
always @ (posedge final_clk or negedge rst_n)
begin
  if (rst_n == 0)
    din_2 <= 1'b0;
  else
    din_2 <= din_1;
end

assign dout = din_2;

endmodule_

module clkgate (
  input   clk_in,
  input   en,
  output  clk_out
);

reg latch /* verilator clock_enable */;

always @*
begin
  if (!clk_in)
    begin
      latch = en;
    end
end

assign clk_out = clk_in & latch;
endmodule

Indeed, both sc_example__DOT__u_clk_gate_1__DOT__latch and sc_example__DOT__u_clk_gate_2__DOT__latch are marked with [aCLKEN] in the log file.

I can't tell why there is no [aCLKEN] for the latch in the 2nd clock_gating in our actual project, it uses the same clkgate module as the example above.

Actually, there are 3 level clock-gating in the project, latches in the first and third clkgate are marked with [aCLKEN], the second one is ignored.

 If this is really the problem, could you point out the way to find out the reason why Verilator misses/ignores the latch in the 2nd clock-gating instance?

wsnyder, 2021-01-13:
Making a small example was the right thing to try.  As that didn't work, you'll unfortunately need to debug the code on the larger design, that is look at where the attribute is correctly and use debug prints or a debugger to trace why it isn't marking the latch appropriately.

zenghan2088, 2021-01-14:
@wsnyder  It Looks like because --no-order-clock-delay is used to build our SystemC model. It has been in our build script for a long time. With --no-order-clock-delay, a "VinpClk" cloned clock variable __VinpClk__TOP__sc_module_top.gate_clk_2 is used in the _eval() function; Without it, _eval() uses sc_module_top.gate_clk_2  directly, then outputs correct waveform.

zenghan2088, 2021-01-14:
@yanx21 comment out "if(clk1) begin" in your second always block,  you can get correct display outputs

wsnyder, 2021-01-14:
Sounds this is solved, -no-order-clock-delay is expected to cause problems ;)

closed 2021-01-14 by wsnyder