verilator/verilator #2562: "-- threads n" with "--hierarchical" slows down simulation time by order of magnitude for n > 1

ayewinoung, 2020-09-21:
Simulation run time dramatically "slow" down by order of magnitude when "--threads n", where n > 1.
I've tried this across multiple CPU vendors on AWS cloud and its consistent across all CPU types from Intel Xeons, AMD EPYC, ARM Graviton2 processors.  I've double-checked its not memory bottleneck by using at least 64Gbyte memory machines for simulation which requires < 1Gbyte. I've also double-checked this across two different designs of similar size.

One data point is, the designs are big enough that, I need to use "--hierarchical" to compile, otherwise it will not compile.  Another data point is, simulation slows down is *not* observed when I cut down the designs so it can compile without "--hierarchical", (i.e. flat compile).

I've kept all other variables constants as I can, and the followings are details :

Ubuntu 20.04 LTS
Verilator 4.101 devel rev v4.100-12-g430238a6
clang version 10.0.0-4ubuntu1 
systemc-2.3.2
 
Any compile switches, tricks that I missed? any pointers much appreciated.

Thanks in advance.

wsnyder, 2020-09-21:
I don't think there's been much analysis of threads interacting with hierarchical.  I suspect each hieararchial piece has its own thread chain, so then the top level requires a context switch at each call down level.

Will require some analysis and improvements.  Maybe @yTakatsukasa or others can take a look?

yTakatsukasa, 2020-09-21:
Hierarchical verilation is very young feature that I could not pay any effort on performance wise.
I am happy to hear that that the feature let you verilate such a big design though performance challenge exists.

I agree with @wsnyder that each hierarchical block may spawn its threads though I have little experience on using `--threads` option.

I think the possible experiment would be:
1. Don't set `--threads` option. (No code change necessary)
2. Set `--threads 1` to hierarhical blocks but the top gets `--threads n` (requires a little code change)
3. Have a global thread pool (Needs research, I even don't know if it is feasible or not)

Could you try not to set `--threads` option first ? (1.above)
It is similar to `--threads 1`, but the verilated result is not thread safe, it means that no overhead for thread-safety.

I can push an experimental branch to try 2. tomorrow or so.

I can not predict the ETA of further improvements.
Maybe we need some open source design that is big enough for this public experiment.

ayewinoung, 2020-09-21:
I've done some experiments between "--threads 1" vs "no --threads", there is no statically meaning full difference between the two, across multiple runs on both Intel & AMD platforms.

Also, anyone interested in improving this issue as commercial contract ?

yTakatsukasa, 2020-09-21:
Thanks for the experiment.

I pushed 2. above to https://github.com/yTakatsukasa/verilator/tree/issue2562_hier_threads that passes `--threads 1` to hier_blocks.
Could you try the branch. (Only 1 commit is added to the latest master).

I hope this will **mitigate** some of overhead you observed.

ayewinoung, 2020-09-21:
I compile the tool and try to run it, got the following error.  I did confirm that going back to release 4.100 does not have this error and can compile and run the sims.

FYI - module : bch_eng_2x2 is the hier_block.

%Error-TIMESCALEMOD: /mnt/projects/CryptoAccel/Sirius/tests/simple_hash/../../..//SystemLink/rtl/src/sys_link_phy.v:16:8: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     obj_dir/Vbch_eng_2x2/bch_eng_2x2.sv:8:8: ... Location of module with timescale
    8 | module bch_eng_2x2 (

yTakatsukasa, 2020-09-23:
PR #2554, that is merged after 4.100 is released, emits timescale if the input module has timescale, timeunit, or timeprecision.

I suspect 4.100 with `--hierarchical` passed by accident and would show the error if `--hierarchical` was not set. (maybe the filelist order affects, not sure).

A preferred solution would be adding timescale/timeunit/timeprecision to files or remove them completely.
The quickest way is probably adding `--timescale 1ns/1ps` option. (`1ns/1ps` should be same value as specified in the HDL files).

ayewinoung, 2020-09-23:
Thanks for the tips, I can compile and run the sims with "--timescale 1ns/1ps" and setting sc_set_time_resoloution(SC_PS).
I have run a few scenarios and general observations:
-  "--threads 1" seems to be working, and only use 1 CPU as specified
- still anything more than 1 in "--threads n" still slowdowns by order of magnitude, and also use as many CPU/Threads available on the machine, (i.e. don't stick to specified n)

Below is a simple comparison of running exactly the same code, on exactly the same machine and the only difference is the compile option. i9-9900K with 64Gbyte Memory.

Verilator 4.101 devel rev v4.022-933-g7a9e79ea

CompileOptions            RunTime (sec)            PeakMemory (Mbytes)              RunTimeCPU Use (Threads)
============          ===========           =================             ====================
no "-- threads"                     169                                     349                                               1
"-- threads 1"                        179                                     349                                               1
"-- threads 2"                      2001                                    391                                              16

Let me know if I can help more to improve "-- threads 2" and above.

yTakatsukasa, 2020-09-23:
Hmm,  could you check *_hierMkArgs.f files in obj_dir ?
I expect only  `V%{top_module}_hierMkArgs.f` has `--threads 2` and the others contain `--threads 1` .
Note that `%{top_module}` means your top module name.

If not, could you retry the verilation after deleting obj_dir ?

ayewinoung, 2020-09-24:
Checked both *_hierMkArgs.f" in obj_dir, they both have "--threads" "2" and I've attached the actual files.
Also, I always delete the obj_dir, between each compile.
Top Module : sirius_tb_hierMkArgs.f
Hierarchical Module : Vbch_eng_2x2_hierMkArgs.f

ayewinoung, 2020-09-24:
Actually attaching files for real..
[sirius_tb_hierMkArgs.f.txt](https://github.com/verilator/verilator/files/5272802/sirius_tb_hierMkArgs.f.txt)
[Vbch_eng_2x2_hierMkArgs.f.txt](https://github.com/verilator/verilator/files/5272803/Vbch_eng_2x2_hierMkArgs.f.txt)

yTakatsukasa, 2020-09-24:
Ah, I am sorry that I missed the version string.
It should be `Verilator 4.101 devel rev v4.100-21-gb3fcd945`.

Please clone from https://github.com/yTakatsukasa/verilator.git then checkout the branch.
> git checkout issue2562_hier_threads

Building the branch is as same as usual.
> ./configure
> make

ayewinoung, 2020-09-24:
Did the check out as requested, "make clean", "./configure", "make -j 8", "sudo make install", still got different version than you suggested.
Missing something here ?

aoung@Shuttle370V2:/mnt/tools/verilator/verilator_try2$ git checkout issue2562_hier_threads
Already on 'issue2562_hier_threads'
Your branch is up to date with 'origin/issue2562_hier_threads'.
aoung@Shuttle370V2:/mnt/tools/verilator/verilator_try2$ git pull
Already up to date.
aoung@Shuttle370V2:/mnt/tools/verilator/verilator_try2$
:
:
aoung@Shuttle370V2:/mnt/tools/verilator/verilator_try2$ verilator --version
Verilator 4.101 devel rev v4.022-934-gb3fcd945

yTakatsukasa, 2020-09-24:
I see. The hash in the last part of version string is `b3fcd945`. So I think you can use it.

ayewinoung, 2020-09-24:
Ok, I compiled with the following version of the tool and still no joy.  
"--threads 2" takes x20 long run time, compared to "--threads 1" and used up all the 16 Threads available. 

Verilator 4.101 devel rev v4.022-934-gb3fcd945

yTakatsukasa, 2020-09-28:
The branch `issue2562_hier_threads` properly limits the total number of threads as expected on my side.
I expect Vbch_eng_2x2_hierMkArgs.f.txt to have `--threads 1` while sirius_tb_hier_MkArgs.f has `--threads 2`.

[sirius_tb_hierMkArgs.f](https://github.com/verilator/verilator/files/5272802/sirius_tb_hierMkArgs.f.txt ) that you uploaded in https://github.com/verilator/verilator/issues/2562#issuecomment-698048480 tells that you are verilating for SystemC testbench.
Is `sirius_tb` instantiated multiple times in your SystemC top module (or sc_main) ?
If so, an instance may run with 2 threads, 8 of instances may run with 16 threads??? (I don't know much about --threads options ).  Then it not likely to be `--hierarchical` specific.

ayewinoung, 2020-09-28:
Is sirius_tb instantiated multiple times in your SystemC top module (or sc_main)?
[Answer] sirius_tb is instantiated only once in sc_main 

I'll try to *only* verilate rtl files and compile the SystemC/C++ files and try to link the two, I've been having trouble doing that.  I'm not expert at Makefile magic, so need to learn that bits.

ayewinoung, 2020-10-01:
I managed to split the RTL files into one verilate/compile pass and the rest of testbench/test SystemC/C++ files into different compile and link them to run.
Still no joy, the same behavior, once I set "--threads n", where n > 1, the simulations slow down by x20 times.
Happy to try experimental builds.

yTakatsukasa, 2020-10-02:
I am sorry that I can not reproduce such too many threads symptom.

I'd like to know the following things.
1. Do you still see many threads more than set via `--threads` ?
2. Does Vbch_eng_2x2_hierMkArgs.f have `--threads 1` reagardless of the option to the top module ?
3. Do you have any other kind of verilated modules in sc_main ? (I know that there is only one sirius_tb instance. Is there any modules other than sirius_tb in sc_main ?)

I am asking 3. because I found the following comment.
According to the comment in the source code, if you are combining multiple verilated modules, there will be more threads though that is not limited to hier_block.
https://github.com/verilator/verilator/blob/5033c906fdedd2c8d11e3c89aad76964191cbb8e/src/V3EmitC.cpp#L2348-L2364

Please remember that the current hier_block is less efficient than flat (i.e. no hier_block) verilation because the top module can not schedule threads in an optimal way as the top module does not know the detail of hierarchical blocks.

ayewinoung, 2020-10-02:
Thanks for getting back, I did another git pull today and re-compile verilator.  It *no* longer used up all the threads when I specify "-- threads 2". It also pass only "--threads 1" to hier_block as you mentioned. 
However, in hier_block compile, "-- threads 2" and above,  *slow* downs run-time by small amount, compare to "--threads 1".
In flat compile, run-time *speed up* somewhat linearly by amount of "-- threads" allocated to it, depending on each CPU arch. 

Verilator 4.101 devel rev v4.022-934-gb3fcd945

closed 2020-10-04 by yTakatsukasa

yTakatsukasa, 2020-10-04:
Thanks for the experiment.
I merged the change to pass `--threads 1` to master.

Further performance improvement is filed in #2583.

PR #2581 (2020-10-04): Fix  performance degradation when --threads is used with --hierarchical (Issue 2562)
Fixes #2562

hier_blocks should be just single threaded otherwise too many threads will be created and the performance may be too bad.

I will merge tomorrow if CI is happy.