verilator/verilator #2563: urandom_range compilation error

nanduraj1, 2020-09-22:
```systemverilog
module t(/*AUTOARG*/);
   int unsigned v1;
   int unsigned v2;

   initial begin
       v1 = 100;
       v2 = $urandom_range(0, v2);
   end
endmodule
```
```
./t_urandom_range.pl 
======================================================================
dist/t_urandom_range: ==================================================
-Skip: dist/t_urandom_range: scenario 'dist' not enabled for test
dist/t_urandom_range: %Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Unsup 0  Time 0:00
======================================================================
vlt/t_urandom_range: ==================================================
        perl /home/ubuntu/softwares/verilator/bin/verilator --prefix Vt_urandom_range ../obj_vlt/t_urandom_range/Vt_urandom_range__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_urandom_range -OD --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_urandom_range t/t_urandom_range.v    > obj_vlt/t_urandom_range/vlt_compile.log
%Error: Internal Error: t/t_urandom_range.v:13:13: ../V3Width.cpp:4734: Node has no type
                                                 : ... In instance t
   13 |        v2 = $urandom_range(0, v2);
      |             ^~~~~~~~~~~~~~
                        ... See the manual and https://verilator.org for more assistance.
%Warning: vlt/t_urandom_range: Exec of perl failed: %Error: Internal Error: t/t_urandom_range.v:13:13: ../V3Width.cpp:4734: Node has no type

vlt/t_urandom_range: %Error: Exec of perl failed: %Error: Internal Error: t/t_urandom_range.v:13:13: ../V3Width.cpp:4734: Node has no type
vlt/t_urandom_range: FAILED: Exec of perl failed: %Error: Internal Error: t/t_urandom_range.v:13:13: ../V3Width.cpp:4734: Node has no type
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00

======================================================================
        #vlt/t_urandom_range: %Error: Exec of perl failed: %Error: Internal Error: t/t_urandom_range.v:13:13: ../V3Width.cpp:4734: Node has no type
                make && test_regress/t/t_urandom_range.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:00

```

wsnyder, 2020-09-22:
Thanks for the report, infant bug.

closed 2020-09-22 by wsnyder

nanduraj1, 2020-09-22:
thanks @wsnyder