verilator/verilator #2566: Support for 'inside' with parameter arrays

dpetrisko, 2020-09-23:
Hello, 

I'd like to request support for using 'inside' with parameter arrays.  Trying the following example

```
module foo
 #(parameter int checklist_p [2:0] = '{0, 1, 2})
 ();

 localparam pass_lp = 1;
 localparam fail_lp = 4;

 initial
   begin
     if (pass_lp inside {checklist_p})
       $display("PASS %d is in checklist_p %p", pass_lp, checklist_p);

     if (fail_lp inside {checklist_p})
       $error("FAIL %d is in checklist_p %p", fail_lp, checklist_p);
   end

 endmodule
```

results in

```
$ verilator --cc foo.v
%Error: foo.v:11: Unsupported: RHS of ==? or !=? must be constant to be synthesizable
%Error: foo.v:14: Unsupported: RHS of ==? or !=? must be constant to be synthesizable
%Error: Exiting due to 2 error(s)
%Error: See the manual and http://www.veripool.org/verilator for more assistance.
%Error: Command Failed /usr/bin/verilator_bin --cc foo.v
```

But the parameter array is constant during elaboration, so I believe this should be synthesizable.  The same example works in synopsys VCS

dpetrisko, 2020-09-23:
Seem related to #821

wsnyder, 2020-09-24:
For small arrays this should be easy to convert into a tree of e.g.

     LHS==array_value[0] || LHS==array_value[1] ...

I say small arrays only because this will be a lot of code if large.

Perhaps you'd like to contribute a pull request to fix this - that is make a test_regress test, then where that error is instead modify the AST tree to make the new AstOr/AstEq tree..

dpetrisko, 2020-09-25:
Thanks for the response! It might be a while before I get time to work on it, but I’ll try to figure something out. 

For anyone encountering the same issue, I worked around it by encoding the array as a packed bit array (works for non-unique parameter lists)

wsnyder, 2023-09-26:
Note recent versions give a clear unsupported warning

`%Error: Unsupported: inside (set membership operator) on unpacked array
`

dpetrisko, 2023-09-26:
Nice!

wsnyder, 2024-08-11:
#4751 added support for this.

closed 2024-08-11 by wsnyder